CTCaer/hekate #696: Failed to mount SD - no way to make hekate see an SD card on Mariko with hwfly

giorgiomassar8, 2022-01-04:
I have installed an hwfly on a Mariko switch. When booting without an SD, hwfly first blinks blue and red for about 2 seconds, then turns green for another 2 seconds and then switches off. Spacecraft-NX screen pops up with the NO SD screen. Till here, everything seems good.

However, when trying to load hekate by putting it as payload.bin on an SD card, it boots to a yellow text screen. with the following text:

Failed to mount SD!
Missing LP0 (sleep mode) lib!

Update boot loader folder!

An exception occurred (LR 40014D01):
UNDEF

Press any key....

If you go ahead you get a red text:

Failed to init SD card.
SD card reader is not properly seated!

Now, to be clear:

1) The SD cards have been partition with MBR
2) The SD cards have been formatted with FAT32 both on MacOS and on Windows 10
3) Tried two Intenso 8 GB SDs, plus one Samsung Evo Plus 256 GB
4) Checked with a microscope the SD connector on the switch mainboard, all pins are perfect
5) Replaced the SD socket board with another Switch's socket board, no luck

Most issues focus on hardware problems of the SD card (connectors), but if it was an SD card issue, it wouldn't even load the first stage payload.bin isn't it?

Any idea what else could be done to try to have it boot Hekate?

Thanks a lot!

urherenow, 2022-01-04:
GitHub is not a help forum. Use GBAtemp, and clarify what version of Atmosphere is on your card as well as version of hekate, whether or not you're using emuMMC, and perhaps the directory structure of your card and contents of your .ini files.

giorgiomassar8, 2022-01-04:
You are right, but to me looks like a bug in Hekate for my specific setup, therefore I reported it here.

Tried both Atmosphere 1.2.4 and 1.2.5. Content of the SD card and INI files according to https://rentry.org/MarikoEmuNAND

Not using emuMMC as I couldn't reach that stage yet.

urherenow, 2022-01-04:
Seriously, use gbatemp. Plenty of people using Mariko with hwfly, so it's not a bug. Takes too much time and space to track down your issue here, as it could be anything from those vague and outdated instructions you're following (on a Mariko, autonogc=1 will do nothing for you but will *probably* cause you trouble in the future, if you use cartridges at all) to an oopsy with your soldering.

CTCaer, 2022-01-04:
hekate is loaded in a corrupted state.
So anything in it might or might not work or if it works it will fail later.
spacecraft-nx uses a 3 year old hekate based sdmmc driver so it might be that (assuming it really is not a hw issue).

closed 2022-01-04 by CTCaer