CTCaer/hekate #698: [Feature request] Dock support for broken screen Switches

xTheDoctah, 2022-01-18:
So, basically i have a switch with a broken screen, and i would like to use hekate, but i cant, because after the hekate's payload injection, if i put the switch into the dock, it doesn't work. Yeah probably the solution would be to replace the screen, but i think that might be a nice feature for every person that has a faulty/broken screen or broken/faulty screen connector and screen connector pads.

impeeza, 2022-01-18:
Do you have auto boot configuration?  For me when I have it I do inject the payload, out in the deck and wait until the TV shows lock screen

xTheDoctah, 2022-01-18:
Into the bootloader/ini folder? If you mean that, the answer is no.  With the autoboot can i see the screen where are shown the various boot options? and the nand dump section ?

impeeza, 2022-01-18:
The auto boot option on the hekate_ipl.ini file a payload is selected automatically and never enter on Nyx boot directly

xTheDoctah, 2022-01-18:
Ok, but in that case i cannot enable patches, right?

impeeza, 2022-01-18:
Yes you can without problem. Booth patches.ini as KIP patches

xTheDoctah, 2022-01-18:
So for instance if i put as a payload if i put as a payload Glee and in the patches.ini the various patch i need, i can run the cfw with the patches?

impeeza, 2022-01-18:
For all effects auto boot is like you tapping a option on nyx

xTheDoctah, 2022-01-18:
So basically the answer to my question is , yes ? xD

CTCaer, 2022-01-20:
As explained many times before, dock support inside hekate will not happen because of the amount of work needed, only to get image showing on tv.

As stated above, better use autoboot and when you need UMS use the boot storage configuration (via homebrew or hex editing the binary)

closed 2022-01-20 by CTCaer

impeeza, 2022-01-20:
Hi, or you can use Pancake Overlay to boot directly to UMS, but after you disconnect your UMS from Computer you will have to reset forcibly the Switch because you are on NYX but no screen ;(  repair the screen is easy.

digitalguy99, 2022-08-10:
> As explained many times before, dock support inside hekate will not happen because of the amount of work needed, only to get image showing on tv.
> 
> As stated above, better use autoboot and when you need UMS use the boot storage configuration (via homebrew or hex editing the binary)

@CTCaer Will you be willing to make it work if I start a fundraiser to pay you for such a project?

DerekRoth, 2023-10-29:
Hello, is there a way to trigger a specific hekate action at boot so I don’t have to go though the menus (as I don’t have a working screen) ? Dumping my nand would be just enough, I think the rest can be done with a PC.

Elektroney, 2025-01-21:
> Hello, is there a way to trigger a specific hekate action at boot so I don’t have to go through the menus (as I don’t have a working screen) ? Dumping my nand would be just enough, I think the rest can be done with a PC.

Anyone?

impeeza, 2025-01-21:
> > Hello, is there a way to trigger a specific hekate action at boot so I don’t have to go through the menus (as I don’t have a working screen) ? Dumping my nand would be just enough, I think the rest can be done with a PC.
> 
> Anyone?

https://github.com/CTCaer/hekate/issues/698#issuecomment-1017442822

please read documentation on README.MD to setup an autoboot config.

Elektroney, 2025-01-21:
I'm sorry but i do not seem to see a way in the readme to dump the nand / create an emunand by setting a key to a value in the autoboot config. Could you point me to the paragraph in the readme

DerekRoth, 2025-03-02:
@Elektroney i figured it out. It's not in the ini file. You actually have to use a hex editor to change the Hekate binary and change some values based on this documentation: https://github.com/CTCaer/hekate/?tab=readme-ov-file#payload-storage
It was a bit unclear to me how this worked but i ended up creating different versions of the Hekate binary for the different use cases. I can send them to you if you want. At the time i was using Hekate 6.0.7.

I have no idea why this is not in the ini file, there is probably a technical reason behind it.

impeeza, 2025-03-02:
You do not need to edit the binaries at all!

you only need the `autoboot` item on your `hekate_ipl.ini`

example:
```
[config]
autoboot=2
autoboot_list=0
bootwait=1
noticker=0
autohosoff=1
autonogc=1
bootprotect=0
updater2p=1
backlight=91

{--- Custom Firmware ---}
{--- SYSNAND START---}
[CFW (SYSNAND)]
id=SYSNAND
emummc_force_disable=1
fss0=atmosphere/package3
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/atmosphere_hue_nobox.bmp
kip1patch=nosigchk
{--- SYSNAND END ---}

{--- EMUNAND START---}
[CFW (EMUMMC)]
id=EMUNAND
emummcforce=1
fss0=atmosphere/package3
atmosphere=1
logopath=bootloader/EnergyStarBL.bmp
icon=bootloader/res/payload_hue_nobox.bmp
kip1patch=nosigchk
usb3force=1

{--- EMUNAND END ---}
{--- Custom Firmware END ---}

{--- STOCK Firmware ---}
{--- Stock START---}
[Stock (SYSNAND)]
id=STOCK
emummc_force_disable=1
fss0=atmosphere/package3
stock=1
icon=bootloader/res/switch_hue_nobox.bmp
{--- Stock END ---}
{--- STOCK Firmware END ---}

{--- PAYLOADS ---}
{--- Atmosphère payload ---}
[Atmosphere fusee.bin+fs (Payload)]
id=Atmosphere
atmosphere=1
payload=bootloader/payloads/fusee_1_08_0.bin
logopath=bootloader/res/atmosphereFuseeBootLogo.bmp
icon=bootloader/res/atmosphereFusee_hue_nobox.bmp

[Atmosphere fusee.binORG (Payload)]
id=Atmosphere
atmosphere=1
payload=bootloader/payloads/fuseeSciresM.bin
logopath=bootloader/res/atmosphereFuseeBootLogo.bmp
icon=bootloader/res/atmosphereFusee_hue_nobox.bmp

{--- PAYLOADS END ---}
```

this Hekate always autoboot to the second item on this case `[CFW (EMUMMC)]` which is EmuNAND .

My console is always docked and when I restart it or cold boot, will always start on my EmuNAND.

impeeza, 2025-03-03:
https://github.com/HookedBehemoth/studious-pancake Ha a reboot to UMS,  may be can reboot to dump NAND.

DerekRoth, 2025-03-03:
In order to have access to this I would need to have installed an emu nand already which defeats the purpose. I don't have a screen and I just need to do the initial sd and emu nand setup of a picofly.

mrDeadLinez, 2026-11-16:
Maybe not actual anymore, but there is a try to do this https://github.com/Retrogendude/Hekate-nova-pro