CTCaer/hekate #699: It seems that hekate does not support ams-1.2.6. What lion tiger will launch a new generation of hekate?

Ic-any, 2022-01-20:
Hello, Nintendo recently updated the system of 13.2.1, and the atmosphere has also been updated to 1.2.6, but it seems that hekate does not support this version of atmosphere. I know that it can be downgraded back to 13.2 with daybreak. I hope you can launch a new generation of hekate as soon as possible。If you use hekate to enter the boot interface and click lunch, you will not be able to enter the virtual system after selecting emummc, and an error will be reported after entering.

Ic-any, 2022-01-20:
Machine translation may have syntax errors, please forgive me

Yoti, 2022-01-20:
Wait for hekate update.

impeeza, 2022-01-20:
You can always fork the repository, make the needed changes and create a pull request; or you can thanks to the developers for they great effort and politely wait for an update

CTCaer, 2022-01-20:
You mean doesn't support HOS 13.2.1.
Update will happen today anyway.

Ic-any, 2022-01-20:
Thank you. We are also in a hurry, but we entered the system with the original boot, but it is not very convenient. We look forward to updating it today.

Ic-any, 2022-01-20:
Maybe because of jet lag, I may have to wait until tomorrow to replace the old hekate, but I really appreciate your reply. bye!

CTCaer, 2022-01-20:
Get latest release that adds support

closed 2022-01-20 by CTCaer

Heath123, 2022-01-26:
> You can always fork the repository, make the needed changes and create a pull request; or you can thanks to the developers for they great effort and politely wait for an update

Or you can open an issue in case the developers aren't aware and to help track progress. I don't see how that's impolite

Masamune3210, 2022-01-26:
Please dont comment on closed issues for no reason, everybody who is subscribed gets a email every comment