CTCaer/hekate #700: Nintendo Switch not starting the payload

ToxicTolerance, 2022-01-22:
Hi,

im using TegraRCMGui 2.6 with the latest Hekate release.
My Nintendo Switch is running on HOS 13.2.1 which should be supported.
The result from TegraRCMGui is always success but the switch is showing nothing.
At the moment i cannot provide more information on this.

Thank you in advance.

impeeza, 2022-01-23:
Hi. Please provide more information about your config, files on SD etc. But I do recommend go to a support forum or channel. This space is for bug report not help and support ask.

masagrator, 2022-01-23:
This doesn't sound like issue with Hekate. What is TegraRCMGui outputting in log window? 0x7000 or 0x0000?

ToxicTolerance, 2022-01-23:
Hey guys sry, my switch is Ipatched, I didn't understood this earlier due to a lack of knowledge. I read a bit in the meantime and understand this now. @masagrator as you might have already gussed, it shows 0x0000. 

Thanks to you guys.

closed 2022-01-23 by ToxicTolerance