CTCaer/hekate #703: [Suggestion] Select emummc from hekate_ipl.ini

josete2k, 2022-01-26:
For pegaswitch users... And for People who don't want to upgrade their sysmmc

It should be a nice feature to Select the emunand files for ams launching.

Something like this:

[emuMMC AMS]
fss0=atmosphere/package3
emummcforce=1
kip1patch=nosigchk
emusource=xxxxx

[emuMMC STOCK]
fss0=atmosphere/package3
emummcforce=1
stock=1
emusource=yyyyyy

Is this possible?

josete2k, 2022-01-26:
Thanks for your reply.

Yes, switching between multiple emunand files is possible... But I thought it could be a good idea to point them from the ipl.ini file in order to avoid unwanted bans.

So you could launch a clean emunand with the stock flag and another emunand with all patches...

urherenow, 2022-01-26:
Sigh... RTFM. Been possible for a very long while now.

 Want an example? https://gbatemp.net/threads/multiple-emummc-setup-including-restoring-a-clean-nand-for-online-use.562188/

josete2k, 2022-01-26:
Ouch... Your right! 

I was looking for this some days. 

Thanks...

closed 2022-01-26 by josete2k