CTCaer/hekate #705: Weird resizing issue

sthetix, 2022-01-27:
So, I just found out that when we resize the emuMMC using Hekate to 12GB, we will have 4.6GB of free space

While doing the same with the NXNANDManager will give us 9.3GB of free space.

How to fix this issue?

Thx

[image]
[image]

impeeza, 2022-01-27:
I think is because you did select the option "format use partition" on NXNANDMANAGER

CTCaer, 2022-01-27:
There's an existing convo here https://github.com/CTCaer/hekate/issues/675 for OLED generally and also that resizing issue.

If possible, I need some log from the nyx.bin I provided there.

closed 2022-01-27 by CTCaer