CTCaer/hekate #707: A fatal error occurred when running atmosphere after updating hekate.

K4llyB, 2022-01-28:
I recently updated my switch to firmware 13.1.2, and once it had turned off I then extracted the latest atmosphere and hekate. 

Once I tried booting into hekate, it gives me the error:
     'failed to init or mount the sd card'. 
After that, I tried to boot into atmosphere but again got an error:
     'A fatal error occurred when running atmosphere.
     Program ID: 010041544d530000
     Error Desc: std::abort called (0xffe)'

I've tried deleting both the atmosphere and bootloader folders and redownloading them. I plugged my sd card into my computer and I can still read and write to it, so I'm convinced that it isn't the sd card that's the problem.

Does anyone else have this issue or have any fixes? Thanks in advance.

Masamune3210, 2022-01-28:
Go back and fill out the issue template instead of just blowing it away

K4llyB, 2022-01-28:
> Go back and fill out the issue template instead of just blowing it away

My bad, this is my first time interacting on Github. Where can I find this template? What am I missing?

CTCaer, 2022-01-28:
@Masamune3210 get your act straight

@K4llyB you probably had something corrupted or sd card issue.
Get a screenshot from Console info -> microsd and up it here

After that fix the archive bits from Tools -> 2nd tab at the bottom -> Fix arc bit and try to boot again

Masamune3210, 2022-01-29:
I formally apologize @K4llyB , I was having a bad day but that is no excuse for that

K4llyB, 2022-01-29:
@Masamune3210 No worries mate. Happens to the best of us.

@CTCaer Thanks for the advice. This is going to sound so stupid, but all I did was take the SD card reader out and put it back in, and it's working again. Might've been just a loose connection. But who knows, this might just be a temporary fix so if it happens again I'll make sure to follow your steps.

I Appreciate all of you.

closed 2022-02-11 by CTCaer