CTCaer/hekate #709: Latest release is not correctly shown in github

Devilmoon, 2022-02-11:
https://github.com/CTCaer/hekate/releases/latest/ redirects to https://github.com/CTCaer/hekate/releases/tag/v5.6.5 instead of https://github.com/CTCaer/hekate/releases/tag/v5.7.0
Going to https://github.com/CTCaer/hekate/releases also shows 5.6.5 as the latest release; the only way to find the actual latest is to manually go to Tags instead of Releases

CTCaer, 2022-02-11:
what the hell
where did it go

urherenow, 2022-02-11:
On my phone, it shows up under all releases...

[image]

CTCaer, 2022-02-11:
It doesn't show up anywhere for me (pc/phone). Only the direct links work.
Anyway, hopefully github will fix that

CTCaer, 2022-02-11:
Seems fixed now, after editing and saving release with no changes.
Thanks for reporting. Reopen if it happens again.

closed 2022-02-11 by CTCaer