CTCaer/hekate #710: Update boot loader

legofan123456, 2022-02-12:
Hello
I've been trying to install Hekate in Oder to install Linux but apon booting I get the error lib missing please update bootloader, I've replaced all files but it's not work and I've reformatted the card several times but to no avail. I'm using tegrarcm to boot to hekate
Thanks a lot

impeeza, 2022-02-12:
Hi for a issue report okease write down which system do you have, injector way, the exact error you have (a screenshot preferred) and steps to reproduce the error.

Seems like your card is fault or have missing files.

You can read the main page readme.

Or go to support sites and read guides about Linux setup.

CTCaer, 2022-02-16:
fix your partitions. if the files are there for real then you have 2 fat partitions and you put them on the second one.
(hekate/nyx mounts the first available FAT partition)

closed 2022-02-16 by CTCaer