CTCaer/hekate #712: [Feature Request] emuMMC on sysMMC

melterx12, 2022-02-22:
Apologies if this was brought up before, but would it be possible to add a feature to create a second partition on the sysMMC and use that as an emuMMC partition of sorts?

with the OLED switch having 64gb I would like to use half the space as a CFW partition leaving the other half stock.

masagrator, 2022-02-22:
Possible? Sure. But issue lays elsewhere. 

emummc idea is a non-interference with nand. Your request goes against this idea. Resizing sysmmc is a red flag for any online activity.

melterx12, 2022-02-22:
> 

so even if the sysNand was resized to match the original switch partition size it would still be flagged?

masagrator, 2022-02-22:
Yes, because they will get info that your unit is OLED.

melterx12, 2022-02-23:
got it, will be closing this issue then. Thanks.

urherenow, 2022-02-23:
I wouldn't be too quick to close until CTCaer chimes in. He may not want to bother, but if emuemmc can be accessed as a file-based thing (which it can), there actually is no reason that the location of said file can't be within the physical emmc in the user partition. This will not show up as an incorrectly sized emmc to OFW, it will simply look like used space. The only question I would have, is if OFW checks emmc for files that shouldn't be there...

melterx12, 2022-02-23:
> 

hmm good point, I will reopen for now then.

CTCaer, 2022-03-03:
Wrong repo. Although have in mind that even if support is added, I won't make a tool to create it because the time needed to make is not worth it.

Please go to: https://github.com/m4xw/emuMMC

closed 2022-03-03 by CTCaer