CTCaer/hekate #713: Guidance on repartitioning and extending emuMMC to 64GB?

SoftwareGuy, 2022-02-23:
I would like a little guidance on re-partitioning or doing a non-destructive way of resizing my microSD (basically removing a partition, and resizing the emuMMC partition).

Currently, on my 256GB microSD on my switch, I have both HOS, emuMMC (RAW) and Android (USER) partitions setup. I want to remove the Android USER partition and add more storage onto the emuMMC partition (apparently we can size up to 64GB now?)

What's the easiest way of going about this? Or would it be better for me to run Hekate's backup emuMMC option so I have a full emuMMC backup, then partition and restore it and somehow resize the emuMMC from there? Will the emuMMC restore be successful if I'm restoring the 32GB backup into a 64GB partition?

impeeza, 2022-02-23:
This space is for reporting Issues or ask for enhancements.  Please go to support sites like GBATemp, Discord or similar for support.

SoftwareGuy, 2022-02-23:
Alright, no worries. Did ask around some outlets but it must have fallen between the cracks. I'll close the ticket, have a good one. 😃

closed 2022-02-23 by SoftwareGuy