CTCaer/hekate #715: Nyx exception when performing specific sequence of actions

v1993, 2022-03-01:
All of those are done with touchscreen (except for the last step)

1. Select "Options"
2. Press "Auto Boot"
3. Pick configuration
4. Press "Save Options"
5. Press "OK!"
6. Press "B" the physical button

This will result in the following error:

```
Nyx exception occurred (LR 81040D0C):
RESET
Press any key...
```

impeeza, 2022-03-01:
Can confirm happen the same to me
[image]

CTCaer, 2022-03-03:
Thanks. That's a null pointer dereference.
Any place elsewhere that this also happens?
(Basically I manually manage where B button acts like back/close/etc. The above case is one that I set it and not clear it when the dialog closes.)

impeeza, 2022-03-03:
That little rascals (null pointers) they hide everywhere.  Thanks for your effort

CTCaer, 2022-03-23:
Fixed in latest version. Thanks for reporting it.

closed 2022-03-23 by CTCaer