CTCaer/hekate #718: UMS in USB Tools not working

ddubs13, 2022-03-10:
I'm trying to use this feature, to read my SD card, and it says it's connected and it's started UMS, but the switch does not show up in my PC drive list at all. I have tried different USB cables and different ports, and there was no change. I've asked the Nintendo Homebrew discord and they had no idea what could be causing it.

urherenow, 2022-03-10:
What about gbatemp? Github is not a support forum, and you haven't provided ANY info that would be required to work through this anyway. Like:

OS (and version of the OS)
File system of your card (fat32? exfat?)
Size of your card
Screenshot of the SD info from within nyx would tell us lots...

ddubs13, 2022-03-10:
Not a support forum? What do you call the issues tab? What is it for?

OS is Windows 64-bit. File system is fat32. Size is 128gb. I don't have a screenshot.

urherenow, 2022-03-10:
The point is to MAKE a screenshot. If you are accessing the USB function, it means you can at least access the card and load nyx. We need to see the sd info as displayed by nyx.

And the issues area is for reporting bugs. If you don't understand how to differentiate between a bug and a personal problem, then github s not the place for you. It's for programmers, and for you to just download the things and read the instructions. Then go to a FORUM if you need help. Or Discord.

ddubs13, 2022-03-10:
Yes I did. And it was only temporary, I had to run that .reg again when it failed to work again. Also, most authors on Github do address issues that crop up with their shit. I can come up with at least a dozen. You have a feature that didn't work for me. I reported a feature that did not work. I would call that a bug, or at the very least, an unintended behavior. And even after it worked, it didn't work permanently, despite the instructions saying it only needed to be ran once.

I was perfectly willing to work with you and get you the info you needed, but since you insist on being a toxic jackass towards someone who did nothing but report an unworking feature, you can figure it out yourself. You have 642 closed issues, most of which are people who simply wanted help installing your fucking app who you turned away. Sorry we can't all be elite haxx0rz.

Masamune3210, 2022-03-10:
The feature works fine, this isn't a issue. There is a difference between a issue and needing help with something and this isn't a forum

lowburger, 2022-03-10:
> You have 642 closed issues, most of which are people who simply wanted help installing your fucking app who you turned away.

Why you talk with such a bad manner? hekate developer is CTCaer who didn't reply for you yet, you only spoke with other user...

urherenow, 2022-03-10:
All authors address issues with the software. The problem here is that there is no issue with the software. I said to get help on a forum. I also said a screenshot if the SD info screen would help troubleshoot. But you are refusing to listen. So WHO'S the jackass?

CTCaer, 2022-03-11:
Have in mind that I don't have an issue with sometimes helping users to fix their PC env. @urherenow 
It's true that issues are for actual issues and not support fixing their PCs, but sometimes there is info that is good to exist.
For example, mounting emmc doesn't show up on explorer because windows does not mount them but the partitions are there.
Also the issues are not for programmers here.. A programmer might even find a way to mitigate an actual issue and never report it, and can stay an issue forever. But a user will report it because he doesn't know how to fix it.

Anyway the whole thing went to floor levels and OP started attacking me indirectly, so issue is closed.

closed 2022-03-11 by CTCaer