CTCaer/hekate #719: "SD has GPT only!"

RiifromAnotherPlanet, 2022-03-10:
I switched my 64gb SD card to a 128gb And been running into endless problems.

Fusee payload wont work ( failed to mount sd card) so I tried with hetake. After a loooot of troubleshooting (formatting sd to fat32, redownloading everything, rewritting hetake ipl.ini) I got Atmosphere to launch but now I get

SD has GPT only!

Found pkg1 ( '20220105094454')

Identified pkg1 and mkey 12

Loaded config and pkg1

CFW mode

Generated keys

No mandatory secmon or warmboot provided!

Failed to launch HOS!

I cant find anyone with the same problem and I'm tired

urherenow, 2022-03-10:
This isn't a support forum, and you formatted it wrong. Just format it within nyx and copy your files back afterwards.

RiifromAnotherPlanet, 2022-03-11:
Thanks so much and I'm sorry, I'll be more careful next time. Should I delete the post?

urherenow, 2022-03-11:
There's no deleting it, but you can close the issue as the author. If allowing nyx to format solved your problem, then yes, close it.

CTCaer, 2022-03-11:
You should use Nyx to format as already suggested. Other than that, post a screenshot or photo from `Console info -> microSD`

RiifromAnotherPlanet, 2022-03-13:
Thanks for all the help. Nyx is the same as hetake right? So I should format my sd card from my switch and not my pc (having a backup on my pc as formatting deletes data. 

[image]
)

Masamune3210, 2022-03-13:
Yes, Nyx is Hekate's GUI and extended tools. If there is a small enough amount of files on the card, the formatter will even backup and replace the files for you, just select to format with the sd card being one contiguous fat32 partition

RiifromAnotherPlanet, 2022-03-13:
Where is that option? What's it's name? I've searched for some formatting tool and I can't find anything by that name

Masamune3210, 2022-03-13:
I believe it's in tools or utilities and named partition SD card or something similar. I'm not near my switch right now, sorry

urherenow, 2022-03-13:
Yes, make sure your files are backed up, and don't forget your emummc backup, if you're using a partition based emummc, then let nyx partition your card.

RiifromAnotherPlanet, 2022-03-14:
Thank you all so much! Really. For someone who was posting the wrong thing everyone was so ready to help. It made me feel so welcomed even though I know so little. Have a good one!

closed 2022-03-14 by RiifromAnotherPlanet