CTCaer/hekate #721: UMS mounting issue

Dola-Shuvi, 2022-03-14:
Good day,
I have recently modded my V1 switch and so far it's working great. The only issue I have run into is using mounting the SD card via UMS, both in hekate as well as TegraRcmGUI. The switch gets detected as a device on my PC but shows up with 0MB storage and doesn't get assigned a driver letter. Diskpart lists the disk correctly but also can not find any partitions. Outside of UMS the SD card works perfectly and is less than a month old.

So far I have only found #685 and #604 which are the exact issue I'm experiencing, however there is no detailed information on how it was eventually fixed. I believe this is some sort of driver issue with Windows but I'm honestly lost on how to fix it as there is nothing being logged in event viewer. If possible I would like to avoid having to wipe my Windows install.

Some additional information on my setup:
Switch is running firmware 13.2.1
SD card is 256GB formatted as FAT32+emummc
Using hekate v5.7.0 & Nyx v1.2.0
CFW is Atmosphère 1.2.6
USB port used is a 3.0 port connected to the motherboard 3.0 header (additionally tested with a 2.0 port integrated into my SD reader)

I would appreciate any help or additional information that  could lead to a solution for this issue.

Thank you in advance for any help and for creating this amazing project.

CTCaer, 2022-03-14:
Well MBR and HW seems fine.

In your case I would check that you didn't replace the UMS drivers with sth else like libusbk.
That's how it should show up:
By connection:
[image]
By type:
[image]

Dola-Shuvi, 2022-03-14:
Looks to me like it loaded the correct drivers properly. I did a test under Debian and it properly mounted there so I assume a hardware issue can be ruled out at this point.
[image]
[image]
[image]

CTCaer, 2022-03-14:
I see you are having too many ext storage readers (probably an all-in-one), did you check the proper driver in disk management?
Because being empty would match the other ones and not hekate UMS.
Also I see you have a filter driver that shouldn't have been there normally. What's fltsrv.sys?
Also send a screenshot from Volumes tab (hit the Populate button first).
[image]

Dola-Shuvi, 2022-03-14:
Correct, the ext storage reader is an all-in-one connected to an internal USB 2.0 header on my mainboard.
The console was connected to a USB 3.0 port on the chassis frontpanel that links to a USB 3.0 header on the mainboard however.
The driver fltsrv.sys is by Acronis and a part of Acronis True Image, a backup software.

Amazingly enough, simply disabling my all-in-one SD reader in device manager somehow allowed my PC to read the SD card via UMS. After disabling and re-enabling my SD reader it now somehow works at the same time.
[image]
[image]

I can't seem to reproduce the issue after this anymore so I'll assume it was Windows being up to weird stuff.

Thank you so much for taking time out of your day to help me troubleshoot this issue. I greatly appreciate it and look forward to the future of this amazing project.

closed 2022-03-14 by Dola-Shuvi

Masamune3210, 2022-03-14:
Likely when you disabled and reenabled the SD reader, it unloaded the filter driver or something else that was conflicting. I would restart and try just to make sure the issue doesn't come back

Dola-Shuvi, 2022-03-14:
I doubt that is the case as even after a reboot UMS and the SD reader both continue to work at the same time.

rockisch, 2022-07-24:
Just for further confirmation, was having the same issue, and I also had an all-in-one SSD reader. Disabling it also fixed the issue for me.

rom4ster, 2023-05-15:
This is unacceptable. Please fix this as this can cause a lot of grief during the setup process. I literally spent 3 hours trying to get this to work.

Masamune3210, 2023-05-15:
Saying things are "unacceptable" in a free piece of software and in a function that is objectively not required to set up a system is probably not the best of ideas. If it's so "unacceptable", then don't accept it and set the SD card up the old fashioned way.

rom4ster, 2023-05-15:
This issue is closed meaning it is refused to be worked on. An issue like this can cause a lot of grief and I think it deserves either to be worked on, or a reasonable explanation on why it cannot be fixed. Closing the issue and basically ignoring people who are having this issue (because it is closed) is probably more mean than saying "unacceptable" if I am honest.

impeeza, 2023-05-15:
Is not an Hekate Issue is your PC OS issue, if you need further support you can go to Discord or GBATemp lots of people look in to that sites than here and you will get help promptly.