CTCaer/hekate #722: [Enhancement] USB Payload Launcher

DraconicNEO, 2022-03-20:
Recently I had an idea for an option to have a USB payload launcher similar to RCM mode that lets you send a payload from a PC or other USB capable device and launch it.

The main reason for this being a feature in Hekate is that on Patched Erista and Mariko systems RCM mode isn't functional. This feature would allow people with a Modchipped Switch to send Payloads over USB just like they would with RCM mode.

urherenow, 2022-03-20:
This is nonsensical. You’re asking this for hekate. Something that needs a mod chip or RCM in the first place, and can launch any payload you choose from the card (and can allow you to add said payload to the card with a usb cable). Think about that for just a minute…

DraconicNEO, 2022-03-20:
> This is nonsensical. You’re asking this for hekate. Something that needs a mod chip or RCM in the first place, and can then launch any payload you choose from the card (and can allow you to add said payload to the card with a usb cable). Think about that for just a minute…

That's a good point, it's too late and I'm too tired. @CTCaer can you please delete this issue after I close it?

closed 2022-03-20 by DraconicNEO