CTCaer/hekate #723: Hekate exception occurred LR 406FC900

JUBYXX, 2022-03-21:
Hello,

after having installed the HWFLY core modchip on my Mariko switch and having put my HATS Pack on my microSD, at startup I come across a very simplistic screen with a few lines:

Failed to init or mount SD!
hekate exception occurred LR 406FC900:
UNDEF

I tried to flash the modchip to update Spacecraft-NX or HWFLY-NX but nothing happens, does the chip have a BGA FPGA IC?

I tried a lot of things, but nothing works, I can't arrive to hekate or something else...

Thanks for your help.

The screen : https://i.ibb.co/7tyDx48/IMG-5533.jpg
My modchip : https://i.ibb.co/16rjnMk/IMG-5534.png

urherenow, 2022-03-21:
If you can't even read the sd card, this is not a hekate problem, and doesn't belong here. Search on gbatemp to find out how (and IF) you can update your mod chip.

Your only solution right now (other than upgrading your chip) is to find an sd card that works with your chip. From there, you *might be able to at least update the sdloader part, using the hwfly-nx tool box.

CTCaer, 2022-03-23:
You problematic ram chips. That artifacting is caused by that

closed 2022-03-23 by CTCaer