CTCaer/hekate #724: Error: "Unknown pkg1 version, HOS version not supported! Failed to launch HOS!"

bennyc023, 2022-03-22:
Hello! After accidentally updating system software while trying to start a game, when trying to launch CFW I get the error code that is in the title. (there are some more similar codes below it but my switch is currently vanilla so I dont have access. I can update with full log if needed)
After scouring the internet, I found many posts on this issue. They all state the fix is updating atmosphere/hetake, but I have the most current versions. I reformatted my SD card multiple times alongside reinstalling the mods, and I am positive I have the most current versions and installed everything correctly.
Did I corrupt my SD card or are hetake/atmosphere out of date for the system update I did? Could it be another issue?
Thanks in advance for the help.

urherenow, 2022-03-22:
A brand new system update was released less than 24 hours ago. Neither Atmosphere nor Hekate have been updated for it yet. You will just have to be patient for a new release.

bennyc023, 2022-03-22:
> A brand new system update was released less than 24 hours ago. Neither Atmosphere nor Hekate have been updated for it yet. You will just have to be patient for a new release.

Thank you. Everytime I have accidentally updated the system software, it was over a week since it originally prompted me so it's always been an easy reinstall. I just wanted to make sure there were no issues on my end because this hasn't happened to me before.
Thank you for the help! I will close this post

closed 2022-03-22 by bennyc023