CTCaer/hekate #726: SysNAND not launching after updating to 14.0.0

LeyendaV, 2022-03-24:
Just updated Hekate from 5.7.0/1.2.0 to 5.7.2/1.2.2, everything was working perfectly fine. After that, I launched my sysNAND and updated it to 14.0.0 from the previous version.
Now, the sysNAND is not launching anymore, screen stays black. Everything else is working fine as usual.

Masamune3210, 2022-03-24:
You likely have sysmodules installed that need updating for 14.0.0. If it makes it into booting HOS it's not a Hekate issue. Try fresh installing atmosphere and I almost guarantee it will boot

LeyendaV, 2022-03-24:
It doesn't show anything at all, it just stays black. My sysNAND is as clean and ut as possible, btw. 
My emuNAND is booting perfectly fine, on the other hand.

impeeza, 2022-03-24:
For support visit discord or GbaTemp,  may be some Sysmodules o bad files.

fernando-granco, 2022-03-24:
Well, I have the exact same issue, emmunand works fine, Sysnand won't boot, it can't be a coincidence. Mine is a Mariko with SX Core.

nosekefik, 2022-03-24:
> Well, I have the exact same issue, emmunand works fine, Sysnand won't boot after update, it can't be a coincidence. Mine is a Mariko with SX Core. Instead of a black screen, it gives me a blue screen forever.

I have same problem here with a Mariko and SX Core.
If I "Reboot to OFW" it starts well, but if I  "Launch stock" I get a blue screen.

fernando-granco, 2022-03-24:
> I have same problem here with a Mariko and SX Core. If I "Reboot to OFW" it starts well, but if I "Launch stock" I get a blue screen.

Oh, just tried the "Reboot to OFW" and it worked, it's a relief. So yes, a problem in Hekate, not the nand files 😄

LeyendaV, 2022-03-25:
> Well, I have the exact same issue, emmunand works fine, Sysnand won't boot after update, it can't be a coincidence. 
> Mine is a Mariko with SX Core. Instead of a black screen, it gives me a blue screen forever. 

Mine is a 2017 Erista. SysNAND is clean, I want to insist on that detail.

LeyendaV, 2022-03-25:
> For support visit discord or GbaTemp,  may be some Sysmodules o bad files.  

The problem is Hekate, since my sysNAND is clean af, and everything else is booting normally (even my emuNAND)

fernando-granco, 2022-03-25:
> The problem is Hekate, since my sysNAND is clean af, and everything else is booting normally (even my emuNAND)

Try reboot to OFW as we did, it's a working solution for the time being

LeyendaV, 2022-03-25:
> > The problem is Hekate, since my sysNAND is clean af, and everything else is booting normally (even my emuNAND)
> 
> Try reboot to OFW as we did, it's a working solution for the time being 

I tried. Same problem, pitch black screen and no sign of anything, I even gave it some minutes to load just in case and still nothing.

urherenow, 2022-03-25:
FFS, your sysnand is NOT CLEAN. Get that through your thick skull. These modchips write to boot 0 when they glitch the system. Fix your boot 0.

LeyendaV, 2022-03-25:
> FFS, your sysnand is NOT CLEAN. Get that through your thick skull. These modchips write to boot 0 when they glitch the system. Fix your boot 0.

I don't have any modchip installed, no idea where or why you are assuming that. My switch is a 2017 Erista with clean sysNAND, and an SD card with file emuNAND, Android and Lakka.

Masamune3210, 2022-03-25:
The other person is using a Mariko, so its mostly to them. Either way I still hesitate to say this is a Hekate issue, if it was outright failing to boot more people would be complaining about the release being broken than just you two

LeyendaV, 2022-03-25:
> The other person is using a Mariko, so its mostly to them. Either way I still hesitate to say this is a Hekate issue, if it was outright failing to boot more people would be complaining about the release being broken than just you two

That's true. Tho it may also be an Hekate issue that manifest only under certain circumstances that we just happen to share.
Tomorrow I'll try installing it again now that my sys in on 14.0.0, maybe updating it after Hekate broke something (it shouldn't, but you never knows)

Masamune3210, 2022-03-25:
Yeah, there is always a chance that there is something weird with your circumstance, maybe CTCaer will have chimed in by then

fernando-granco, 2022-03-25:
We can boot OFW using the "Reboot to OFW" feature, it clearly shows the modules are fine and not broken. SX core discord have plenty of people with the same problem, not everyone comes to github.

LeyendaV, 2022-03-25:
> We can boot OFW using the "Reboot to OFW" feature, it clearly shows the modules are fine and not broken. SX core discord have plenty of people with the same problem, not everyone comes to github.

Reboot to OFW is not working for me, so we may have a similar problem but not exactly the same one.
Also, it's also truth that most people with problems don't come here to ask for support, but rather look for help on several other places.

urherenow, 2022-03-25:
Do you have autoRCM on? "Reboot to OFW" should take hekate completely out of the picture. Since you're on erista, completely power off your switch (hold power for 12 seconds), then turn it on without a jig and without an SD card.

Masamune3210, 2022-03-25:
I have a feeling these two issues are fundamentally different between your twos' systems. A modchip changes quite a bit, and the divergence in symptoms is a sign as well

urherenow, 2022-03-25:
Ok, now that I'm on a computer and can type:

Those with modchips/Mariko, you shouldn't be jumping in on here, as you have a different issue. Your issue, however seems like a legit bug, so one of you should open a new issue with all of the details.

@LeyendaV Sorry if it sounded like I exploded directly at you. TBH, I was getting confused with all of the different posts from different situations. You in particular need troubleshooting and should bring this into a discord server, or at the very least, start a thread at gbatemp, and I'm sure we'll figure out where the problem is (eventually). But for now, you should really close this issue as it's already too long and github isn't meant to be a support forum. Whether you decide to use gbatemp or discord, throw in the server name or link to the github thread when you close the issue. You can always open a new one with better info it is determined to be a bug.

CTCaer, 2022-03-25:
Stock works fine in 14.0.0 for both erista and mariko.
As are SYS CFW and EMU CFW via hekate.

Any panic screen (yellow/blue/black) is because you didn't update atmosphere.
Reminder that exosphere.bin is used on stock as it replaces secmon and in erista warmboot.bin is used also.
(Everything else that is the actual cfw is disabled and not loaded).

So when keys change, don't expect the old exosphere to work. You will just get a black screen or yellow or blue.

Any other issue with updated hekate and atmosphere in CFW environment (stock is always fine because it doesn't load cfw stuff), is because of incompatible sysmodules or mixed atmosphere files.

Finally, don't expect reboot to ofw to work on erista with autorcm enabled.

closed 2022-03-25 by CTCaer

LeyendaV, 2022-03-25:
> Stock works fine in 14.0.0 for both erista and mariko.
> As are SYS CFW and EMU CFW via hekate.
>
> Any panic screen (yellow/blue/black) is because you didn't update atmosphere.
> Reminder that [...]

My sysNAND is completely untouched. I'm not getting panic screens, just black after attempting to launch (only in sysNAND). I'm not using Atmosphere over it nor any other CFW, no need for exosphere or incompatible sysmodules check since I don't have them.

As a side note, Reboot to OFW was working flawlessly till before I updated sys to 14.0.0, and auto RCM was enabled.

urherenow, 2022-03-25:
Now the problem is clear. You have no idea what you're talking about or what you're doing. You can't have tested emummc or cfw of any kind without Atmosphere (and thus, exosphere and such). Hekate, in and of itself, is NOT A CFW.

This issue is closed, so if you really want help, do as I said. Pop into discord or gbatemp.

LeyendaV, 2022-03-25:
> Now the problem is clear. You have no idea what you're talking about or what you're doing. You can't have tested emummc or cfw of any kind without Atmosphere (and thus, exosphere and such). Hekate, [...]

And you are an asshole. We come here to help each other and come up with clear answers for people that may come looking for a fix for a similar problem.
If you are gonna act smart and cocky instead of helping, just go away, nobody likes people like you 🤷🏻‍♀️

Masamune3210, 2022-03-25:
If you say things that dont make sense, don't be surprised when people get annoyed.....

CTCaer, 2022-03-25:
@LeyendaV reread what I wrote.
Again you want it or not, exosphere is used in stock. Update atmosphere also.

Also reboot to OFW doesn't work with autorcm in erista, period.
I break all 4 BCTs when you enable autorcm, so it's impossible to do a normal reboot. That's the point of autorcm.
Of course an update from stock mode, will restore the BCTs and effectively disable autorcm.
Just in case you are confusing why it works after an update.
It's easy to see if it's on, in `Tools -> 2nd tab -> Autorcm`.

And because all boot modes are always tested before a release to see if they are fine, and they are,
in your case I would make sure to update stuff again.

PS: Such issues from users always show up when there's a new update that changes master key.
But as you can see from previous releases, there's no hekate hotfix. Because what I push is first tested and works.
And users eventually learn the process and update everything properly.
Also, after you fix that, keep the whole thing in mind, because it's gonna happen again on the next major HOS update that will change master key. Although if it's a 2017 erista, you should have known that by now.

PS.2: have also in mind that even if stock uses exosphere.bin, that's **not** identifiable by nintendo.
They can't identify it because if they try and use the extensions, the console will panic on OFW.
So imagine having tens of millions of users in OFW getting panics.

LeyendaV, 2022-03-25:
> If you say things that dont make sense, don't be surprised when people get annoyed.....

Honestly, only trash gets annoyed when someone with a problem asks for help.
Obviously things won't make sense in some cases. If they always were, everyone of us would fix their own problems instantly without asking for help. Heck, none of us would have problems to begin with 🤷🏻‍♀️

CTCaer, 2022-03-25:
> FFS, your sysnand is NOT CLEAN. Get that through your thick skull. These modchips write to boot 0 when they glitch the system. Fix your boot 0.

And btw @urherenow you should talk better. I know that you many times try to help, but such behavior that escalates stuff is not accepted here. Worse even, it makes user attack relevant software and become stubborn.
And that's not the first time.

Same thing goes to anyone else that will try and shame a user that doesn't have the knowledge of how some stuff work.

@Masamune3210 
Same thing, don't escalate stuff.

@LeyendaV 
Same also, replying to such comes full circle and escalates even more.

urherenow, 2022-03-25:
> > Now the problem is clear. You have no idea what you're talking about or what you're doing. You can't have tested emummc or cfw of any kind without Atmosphere (and thus, exosphere and such). Hekate, in and of itself, is NOT A CFW.
> > 
> > This issue is closed, so if you really want help, do as I said. Pop into discord or gbatemp.
> 
> And you are an asshole. We come here to help each other and try to find clear answers, also for people that may be looking for a fix for a similar problem.
> If you are gonna act smart and cocky instead of helping, just go away, nobody likes people like you 🤷🏻‍♀️

Wrong answer. People go to forums for help. People come here to report legitimate bugs. You have not only ignored me, but you're also ignoring the author of this software. I've offered real help, if you would only open a gbatemp thread or pop into discord, and I'm quite sure I could have had you sorted in 30 minutes or kess on discord.

But now, you can just go fuck yourself and stay broke, simply because you're stupid.

urherenow, 2022-03-26:
Normally won't continue posting to a closed issue, but I feel like this was unresolved. I just helped another user on temp (Verifying again, that it works fine with a mod chip/Mariko). The root of the problem in that case was a stupid Windows security thing.

Right click on both hekate_ctcaer_5.7.2_Nyx_1.2.2.zip and atmosphere-1.3.0-prerelease-b69fa1357+hbl-2.4.1+hbmenu-3.5.0.zip, and select properties... Is there a check box near the bottom that says "unblock"? If so, unblock it. Then extract both to your card again.

ragiovan, 2024-05-31:
Also, after a lot off this bullcrap answers of bullying from people who pretend to be smart, but `astonishingly` don't give any possible solutions (What's the matter with some of you guys, it's almost like the github community turned into a useless stackoverflow community?), here is one possible solution:

[image]

LeyendaV, 2024-05-31:
> Also, after a lot of this bullcrap answers of bullying from people who pretend to be smart, but `astonishingly` don't give any possible solutions (What's the matter with some of you guys, it's almost like the github community turned into a useless stackoverflow community?)

I feel like this quote should be officially and permanently added to the GitHub home page as a top banner.

impeeza, 2024-05-31:
Guys, this is not a support channel, this is a space for reporting ISSUES AND BUGS on the Hekate.

for support please go to sites like GBATemp, Discord or similar.

There will be lots and lots of people willing to help.

ragiovan, 2024-06-02:
This seemed like an issue or bug to me if I'm not mistaken. If a user has to find a backdoor solution due to a confusing error, it points to an issue for me. 

Now there could be something that's user error that shouldn't be fixed in code, but if that's true at least more relevant error messages could be output.

CTCaer, 2024-06-02:
Hijacking a 2 year old issue, which is not even a hekate related one (should be at atmosphere's repo), is not ok.
If that behavior of bickering or remembering to do that after years happens in other issues, relevant users will be banned.
Unlike what *some* believe, github issues is not for debates/rants or random support like "stackoverflow" or gbatemp.