CTCaer/hekate #727: Can't boot at all

cbr1201, 2022-03-27:
Cannot boot to sysnand or emunand. Using latest verisons of hekate and atmos. Firmware I think is 11.0 because thats the only nand backup I have on me. I have a 13.x one if needed. SD is good and checked with h2testw.
ERROR when I boot emunand/sysnand with logs on

`Initializing...
Found pkg1 ('20210805123730').

Identified pkg1 and mkey 12

Found fss/pk3, Atmosphere 1.2.6-173D5C2D
Max HOS: 13.2.1
Unpacking.. Done!
Loaded configs and pkg1
CFW mode
generated keys
loaded warmboot and secmon
read pkg2
pkg2 decryption failed!
pkg1/pkg2 mismatch or old hekate

failed to launch HOS!`

Please ask for any needed files or anything I can do to help get this solved. THank you

CTCaer, 2022-03-27:
Your pkg1 in BOOT0/1 is 13.0.0 - 13.2.0 while as you claim your pkg2 is 11.0.0

Normally that message is always because you have a pkg1 that does not match a pkg2.
You probably restored GPP without restoring BOOT)/1 also.

Either restore GPP to 13.0.0 - 13.2.0 or restore BOOT0/1 to 11.0.0

closed 2022-03-27 by CTCaer

cbr1201, 2022-03-28:
thank you, sorry im stupid lmfao