CTCaer/hekate #729: blue screen when boot

chenxuuu, 2022-03-28:
after I upgrade to hekate 5.7.2, I got a message `failed to match warmboot with fuses-Sleep Mode won't work`

my device:
83 - Iowa(Mariko) - Retail
sx core with custom boot
Hekate 5.7.2
CFW 11.0.0 emuMMC Atmosphère 0.20.1
OFW 14.0.0 (I used `Genuine boot` in sx os, so...)

I try to do with https://github.com/CTCaer/hekate/issues/659, but I got a blue screen. then no more message, it only blue screen( both ofw and cfw)

```log
pgkg1 ('20220209100019')
Burnt Fuses (ODM 7/6) 16-0 (HOS: 13.2.1+)
ODM Fields (4,6,7) 00010A40 00000000 0000FFFF
```

thanks!

urherenow, 2022-03-28:
And why is Atmosphere 0.20.1?

chenxuuu, 2022-03-28:
> And why is Atmosphere 0.20.1?

I never upgrade atmosphere😂 should I upgrade it?

urherenow, 2022-03-28:
Every. Time.

chenxuuu, 2022-03-28:
I try atmosphere 1.2.6, it's same

urherenow, 2022-03-28:
You should have a folder warmboot_mariko.

Copy that to your PC, then delete it from your card. Then launch "stock".

chenxuuu, 2022-03-28:
@urherenow thsnks

warmboot_mariko folder is there

I backup and delete it, but unfortunately, it still give me a blue screen after I tap 'stock'

urherenow, 2022-03-28:
You didn't update atmosphere. Maybe you tried...

Right-click on the .zip file and check for a security thing at the bottom of the properties. Unblock it. Then extract it to your card again.

CTCaer, 2022-03-28:
You need latest atmosphere files to boot 14.0.0.
Because it's pre-release, most users do not see it and dl 1.2.6.
1.3.0 is the latest and the only one that supports 14.0.0. You need to press the `releases` link.

A blue/yellow/black screen can happen if hekate is latest and supports HOS version and generates proper keys but exosphere from atmosphere does not support them because it's old.
(other reasons for black screen is mixing atmosphere files, like using old kips or sysmodules)

>Right-click on the .zip file and check for a security thing at the bottom of the properties. Unblock it. Then extract it to your card again.

This has nothing to do with that.
hekate and HOS are not windows. The unblock thing only affects Windows and NTFS file systems.
I would be very surprised if the managed to mess that up and broke extraction of random files.
(But with windows vista 11 out there, you never know)
Anyway, that's supposed to protect against running random executables (.exe/.dll).

closed 2022-03-28 by CTCaer

urherenow, 2022-03-28:
Fixed someone on temp 2 days ago with that issue. Package3 wasn't actually being extracted because it was blocked.

chenxuuu, 2022-03-28:
@CTCaer thanks, atmosphere 1.3.0 works

I thought the old cfw system didn't care if it used the old version atmosphere, I'm wrong

impeeza, 2022-03-28:
By the way the "blocked" file on windows, DO NOT avoid to decompress anything from the zip file, only "marks" the file on his metadata as "no trusted, downloaded from internet" as stated by CTCaer that only affect the file on NTFS because the mark is on the metadata streams of that filesystem.  I haven't able to reproduce the "extracting" issue, trying on Windows 11, Windows 10, using winzip, 7zip and explorer to extract the files the package3 file always get extracted the issue could be windows nagging about the "untrusted" status of the files when copying to other disk.

urherenow, 2022-03-28:
https://gbatemp.net/threads/atmosphere-switch-cfw-updated-to-version-1-3-0-adds-ofw-14-0-0-support.609802/post-9785666

Eonfuzz, 2025-07-20:
> You need latest atmosphere files to boot 14.0.0. Because it's pre-release, most users do not see it and dl 1.2.6. **1.3.0** is the latest and the only one that supports 14.0.0. You need to press the [...]

3 years later your post saved me from a blue screen issue my side, thanks mate.