CTCaer/hekate #731: Bootprotect setting causes issue with FTPD

ToxiClay, 2022-03-31:
Version: HOS 13.2.1 | AMS 1.2.6 & 1.3.0

Enabling the bootprotect setting within `hekate_ipl.ini` file causes FTPD and, potentially, all similar homebrew to throw a 550 I/O error and die during the directory retrieval stage due to blocking reading within HOS.

[image]
[image]

See, for example, [this issue](https://github.com/mtheall/ftpd/issues/139) as well as [this issue](https://github.com/mtheall/ftpd/issues/142).

Is there a way to make this setting compatible, or is that a doomed prospect?

impeeza, 2022-03-31:
That is a Homebrew issue, the homebrew should be prepared to a "exception" caused by access denied.  on ftpd-light there is already a issue created: "https://github.com/cathery/sys-ftpd-light/issues/19"

CTCaer, 2022-03-31:
That's a homebrew issue yeah.
Every single hb should check its result codes and manage these specific files (because these folders show as files).

It's not normal in 2022 after 4 years of fighting archive bits issues, for homebrew to not handle these.
Because that means their software will break instantly if user uses Mac OS or copies folders to pc and back, if you don't run the archive bit fixer in hekate.

Anyway, dunno why that issue was opened here, when bootprotect does exactly what it says it does.
Disallows access to HOS on the bootloader folder.

closed 2022-03-31 by CTCaer

ToxiClay, 2022-03-31:
Apologies.

>Anyway, dunno why that issue was opened here, when bootprotect does exactly what it says it does.

I decided to open an issue here because there was already an issue opened under several FTPD repos, and I didn't see one here. I couldn't tell if you were tracking this and what the plan may or may not have been. I'll keep needling the other developers, then.