CTCaer/hekate #732: Blue Screen when trying to launch OFW

Knightzeth, 2022-04-03:
I've never had issues using Hekate on my Mariko Unit with SX core, but after updating to 14.0.0 my OFW, and updating my Hekate to the last version, I'm getting a blue screen when I try to launch the OFW, no problems with CFW just OFW, I'm assuming this must be a problem with Hekate, since I've tried Genuine Boot on my SX Boot, and it launches my 14.0.0 stock FW, I've tried everything with no luck, maybe you can help me find out what's causing this, or maybe is just a bug?

CTCaer, 2022-04-03:
Update atmosphere to 1.3.0 pre-release if you want to launch **Stock**. exosphere is required from it. Otherwise use the reboot to OFW option.

closed 2022-04-03 by CTCaer

Knightzeth, 2022-04-03:
Thank you, for some retarded reason, I was downloading atmosphere 1.2 instead of 1.3 your statement made me double check and understand my problem, thanks again.

imax9000, 2023-10-14:
> Update atmosphere to 1.3.0 pre-release if you want to launch Stock. exosphere is required from it. Otherwise use the reboot to OFW option.

@CTCaer, I think it's worth to add this to FAQ, I almost opened a bug about getting blue screen on sysMMC.

From a user perspective, it's unexpected that `Stock` boot option still uses anything from CFW, and that `Reboot` -> `OFW` option is distinctly different from `Stock`.