CTCaer/hekate #733: Archive Bit Nyx exception occurred

Calebcoolguy1, 2022-04-04:
While trying to perform Fix Archive Bit (all folders) the screen froze at this point. I've tried about 3 times to get past this point.
[image]
On the 3rd attempt, I pressed the close button at the top right and got this message
[image]
(It says specifically: Nyx exception occurred <LR 81000E84> DABRT)
I'm assuming that this might have something to do with the freezing of the Archive Bit but i'm not entirely sure.

BeastMiner32, 2022-04-04:
I have also been getting a similar issue while trying to resolve an issue i was having with Smash Ultimate. Mine has a slightly different code with (LR 81000E14) DABRT and the error shows up without interaction. I am going to format my sdcard and check if it works any better then. But i will keep a backup of this particular version of this.

urherenow, 2022-04-04:
Check your sd info screen for a 1bit width and for a suspect serial number. Also for storage space. If those look good, run h2testw on your card (use all available space).

Calebcoolguy1, 2022-04-04:
I'm not totally sure where to find the information on the 1bit width or suspect serial number, is any of that here or am I on the wrong screen.
[image]

CTCaer, 2022-04-04:
@Calebcoolguy1 that has nothing to do with 1bit more or anything. 99.9% it's corruption.
Especially when you use exfat.
In your case I would do a check and repair of it with chkdsk.
And reply with results of chkdsk and a rerun of archive bit fixer.

I guess I will put some protections against such hangs and and inform user for possible corruption.

Calebcoolguy1, 2022-04-05:
Yeah I switched from exfat to FAT32 and now fix archive bit works as normal, thanks for the help!

closed 2022-04-05 by Calebcoolguy1