CTCaer/hekate #734: [Enhancement] Output to TV when Docked

billyea, 2022-04-04:
Allow Hekate to display through to the TV when the Switch is docked (and presumably accept inputs from wireless controllers).

I regularly jump between Android TV and HOS when my Switch is docked, but Hekate can only have one auto-boot target and there's no way to select whether to boot Android or HOS when the Switch is docked. I need to take the Switch out of the dock and manually select which one to boot.

CTCaer, 2022-04-04:
duplicate of https://github.com/CTCaer/hekate/issues/698

As always, I need to make one million drivers for that to work only to save 10s raising from the chair. Use the boot storage to select where to reboot. Make a tasker script that sets the ID to your HOS config and triggers and normal android reboot.

closed 2022-04-04 by CTCaer

digitalguy99, 2022-08-10:
Will you be willing to make it work if I start a fundraiser to pay you for such a project?

seracohw, 2022-08-31:
HUH?

I KNOW you are tired of explaining this over and over and over for the 100th1,000,000th time or so.

You say to "save 10 seconds of getting up"? The 10 seconds is not the problem people care about, it is the number of times someone has to get up to switch back and forth.

"Use the boot storage to select where to reboot. Make a tasker script that sets the ID to your HOS config and triggers and normal android reboot."

HUH? can we bother you to explain this in basic (kind of dumb person) English?

seracohw, 2022-08-31:
> I'll take a crack at it. It takes more effort than it's ultimately worth to get dock output working during early boot, considering everything has to be handled manually since Horizon OS isn't booted

I can understand the idea and time requirement behind hand written custom drivers. 

That is not what I am asking for, I just need a simpler explanation to his posted alternative. 

I can move from HOS to hekate to Lineage / Ubuntu with nothing more than a blank screen until Lineage finishes booting. I imagine because I did not write the Lineage / Ubuntu boot lines. 

For whatever reason, trying to go from Lineage to hekate to HOS via a “Launch” or “More Configs” turns my dock off? COMPLETELY, showing me no picture even after HOS / Atmosphere has finished booting up. Also, something I noticed last night, it appears to reboot? hekate (shows the hekate screen) before trying to boot Atmosphere. 

Even from inside of hekate, it shows the hekate screen again when I click on my Atomosphere boot option. 

I understand the “Payloads” is doing its own thing and not really using hekate to do anything, but hekate’s Auto Boot feature does not allow you to use the “Payloads” as a boot option, so I had to figure out how to hand write the hekate ipl ini lines to be able to Auto Boot Atmosphere coming from Lineage / Ubuntu. 

I am assuming I have done my boot option wrong somehow because as I said, my dock turns off? completely, showing me nothing, even after Atmosphere has finished booting up, requiring me to redock the switch to get picture again, which was the same thing I was having to do without having the Auto Boot active. 

If what he is telling the other guy to do is suppose to be an alternative to the no tv out, then I have done something wrong.

Mustardnote3943, 2025-08-04:
My nintendo switch screen is unseeable and i wish this feature came.

Masamune3210, 2025-08-07:
Please don't comment on long closed issue reports