CTCaer/hekate #737: [Feature Request] Add support for emummc options when using payload chainloading.

a31user, 2022-04-12:
I would like to request you for emummc options when using payload chainloading. Emummc options such as emummcforce=1 works well on fss0, but won't work on payload chainloading. If people use sxos or reinx which won't work with fss0, emummc options won't work with downgraded emummc.

urherenow, 2022-04-12:
Your quarrel is with reinx and sx, not hekate. SX is dead, so good luck with that. Not that you could ever communicate with the author anyway.

fss0 has nothing to do with either or with emummc anyway.

urherenow, 2022-04-12:
It has N O T H I N G to do with hekate. If you load a payload, NO hekate options matter at all. It's up to that payload to support emummc. Stop bugging the hekate dev.

CTCaer, 2022-04-12:
Piracy cfw discussion is not allowed

closed 2022-04-12 by CTCaer