CTCaer/hekate #738: Error while loading fss0: atmosphere/package3

pastelaura, 2022-04-17:
I am trying to update my Switch's CFW and I keep getting this error. I have tried just about everything to fix it and nothing is working.

I have:

Updated the Atmosphere files on my SD card
Updated the Sigpatches on my SD card
Updated Hekate on my SD card

Updated hekate_ipl.ini to the following:

[config]
autoboot=0
autoboot_list=0
bootwait=0
verification=1
backlight=100
autohosoff=0
autonogc=1
updater2p=0

[Atmosphere]
fss0=atmosphere/package3
kip1patch=nosigchk
atmosphere=1
emummc_force_disable=1

[Atmosphere_emummc]
fss0=atmosphere/package3
kip1patch=nosigchk
atmosphere=1
emummcforce=1

[Original]
fss0=atmosphere/package3
stock=1
emummc_force_disable=1

And freshly installed everything.
After all of that, I still get the error "fss0:atmosphere/package3" and it has started to leave me frustrated. I could really use some help.

Masamune3210, 2022-04-17:
Try checking your card for errors and making sure it isn't faulty.

pastelaura, 2022-04-17:
> Try checking your card for errors and making sure it isn't faulty.

It definitely isn't my card at fault. My PC can read it just fine and it was working perfectly on my Switch yesterday on an older version of Atmosphere/Hekate.

urherenow, 2022-04-17:
What did you “update” with? As in, where did the files come from?

pastelaura, 2022-04-17:
> What did you “update” with? As in, where did the files come from?
> 
> delete atmosphere/package3 then replace it from the atmosphere zip file.

Atmosphere. Got the files from Atmosphere's github page. 

I already did that several times.

impeeza, 2022-04-17:
Please for support go to the proper channel, like discord or GBATemp, this space is for bug report.

your problem is on the atmosphère folder, clear that folder, AND SEPT folder if you have it on the root of your SD.

CTCaer, 2022-04-18:
> > What did you “update” with? As in, where did the files come from?
> > delete atmosphere/package3 then replace it from the atmosphere zip file.
> 
> Atmosphere. Got the files from Atmosphere's github page.
> 
> I already did that several times.

Make sure you used the release zip and not the source zip.
If you did, then it's sd corruption. Delete the atmosphere folder completely and add a fresh one.

That error can only happen if the file is not there or it can't be read.

closed 2022-04-18 by CTCaer