CTCaer/hekate #739: Option to turn gray icon square on/off

BeckysFootSlave, 2022-04-20:
A friend and myself have a project together!
Started with a vertical nxtheme using a switch Cardrige as game icons!

He suggested to also use that Cardrige for hekate 😉
Here is a preview of how it looks:
[image]

Now we ask you if we can get rid of the gray square or a on/off function because the icons have correct alpha channel settings to just show the Cardrige!

Another idea I had in mind was a new icon size for this icons 😬
Still 192 in height but just 128 in width to get 5 icons aligned in one row!
But that just came to my mind!
I can't code and that is the reason I never know how hard it really is to implement such things!

I thank you so much for your time and one of the greatest (of not the greatest) bootloader I have ever seen!

Dankal87, 2022-04-20:
Le dot

urherenow, 2022-04-20:
Couple of suggestions for your icons: OFW is only done with the reboot -> OFW option, so an icon for that is not useful. It should be labeled "stock" instead of OFW, because it's not exactly the same thing. Atmosphere  (payload=bootloader/payloads/fusee.bin) is emuEMMC, if you have one and it is active, yet will launch CFW emmc if you don't have one... however, with fss0, you can choose to launch CFW in emmc OR emuEMMC, so there should be 2 different icons.

Have you posted these icons to gbatemp? Please do so, if not.

BeckysFootSlave, 2022-04-20:
> Couple of suggestions for your icons: OFW is only done with the reboot -> OFW option, so an icon for that is not useful. It should be labeled "stock" instead of OFW, because it's not exactly the [...]

That is a picture of my friend's hekate!
But thank you for mentioning!
Tbh I have also named it OFW (sysMMC) = (Stock) 😬 but I use two different for atmosphere!
[image]
[image]
[image]
[image]
[image]
[image]

These are mine and I also use a different background (but the same frame (frame is made by me)
I will tell him that you said to bring them to Gbatemp! 😉

BeckysFootSlave, 2022-04-30:
Looks really good without border! 
Next test is with a 2x5 grid! 
[image]

BeckysFootSlave, 2022-05-10:
Also the 5x2 grid works like a charm!
No problems encountered while using it!
Just awesome... Speechless! 

[image]

impeeza, 2022-07-21:
Hello there, I don't know if comment here or is better to open a new isse.

on Hekate 5.8.0 if you use a file icon with "HUE" format, it is displayed well and with the color of interface but if THE SAME file is renamed `*_nobox.bmp` then the hue color isn't used the icon is show without the gray background but white.

[image]

on this example all icons are the special ones with "Hue" format, but the second one is named `Payload_nobox.bmp` I don't know if is the wanted behavior or a bug introduced with the new functionalities.

BeckysFootSlave, 2022-07-21:
I don't think that you can mix them up!

And if you can do that = just "nobox" doesn't know that the image is hue and should change the color!
It would be possible if the ending is = icon_hue_nobox.bmp
(don't know if that works?!

But just then nyx would know it is a hue icon with color changing and without the box!

impeeza, 2022-07-21:
Yep, in fact the file currently is called: SD:\bootloader\res\payload_hue_nobox.bmp

alimazharali, 2023-06-28:
> Looks really good without border! Next test is with a 2x5 grid! [image]

please can you give a link to your icons or theme ?

BeckysFootSlave, 2023-06-28:
@alimazharali
It's all selfmade 😘
I can upload them if you want and give you a link! 👍

alimazharali, 2023-06-28:
> @alimazharali It's all selfmade 😘 I can upload them if you want and give you a link! 👍

Excellent work ,.,., most beautiful icons i ever see ,.,.
link plssssssssssssssssss god bless you 
waiting :)

alimazharali, 2023-06-28:
> @alimazharali It's all selfmade 😘 I can upload them if you want and give you a link! 👍

plssss send me all the icons if you can or full theme of hakete as you shown i the thread i am waiting desperately,.,. 
my email is 
alimazharali@yahoo.com

Masamune3210, 2023-06-28:
Be careful about just posting your email in plain text. Bots crawl the internet looking for them so that they can be added to spam lists, not to mention the fact that literally anyone who looks at the comment chain or is subscribed to it has your email now

BeckysFootSlave, 2023-06-28:
Yeah that's right what @Masamune3210 says!

I'll give link to hoster or github link

alimazharali, 2023-06-28:
> Yeah that's right what @Masamune3210 says!
> 
> I'll give link to hoster or github link

pls notify here ,,.,. i will wait ,.,.
millions of thanks ,.,. all hekate icons pls and more if you have

impeeza, 2023-06-28:
Hello there,  all of you if you go to GBATemp.net you will find a thread about how to create that icons, the author has shared them on the thread and you will find also the new set of naming needed for they to work.

BeckysFootSlave, 2023-06-28:
Here the background with and without the hands and all my icons!
I'm going to make new ones and can also post them here!

[bg_and_icons.zip](https://github.com/CTCaer/hekate/files/11897442/bg_and_icons.zip)

Jbb08, 2024-07-02:
@impeeza any chance you can point me to your green icons, I really want my icons to match the main Hekate theme, but for the life of me I can’t seem to find any, yours are the closest I can find, sorry for messaging on a closed threat :-)

urherenow, 2024-07-03:
GitHub isn’t a support forum. If you don’t have any kind of icon entry in hekate_ipl.ini, then you’re already using “the main Hekate theme”.

impeeza, 2024-07-03:
@Jbb08 They are not green they are «Hue» icons, the green color is the color set on NYX,  if you search me on GBATemp.net I will try to help you.