CTCaer/hekate #740: L3 + HOME does not end USB Gamepad emulation

v1993, 2022-04-22:
USB Gamepad pop-up tells that removing cable or pressing L3 + HOME will end HID emulation. While removing the cable indeed works, L3 + HOME does nothing.

CTCaer, 2022-04-26:
Just tried it and it's fine.
Can you test in HOS button test that L3 and HOME works for you?

v1993, 2022-04-26:
Yes, both L3 (pressing down the left stick, right?) and HOME button presses register just fine.

v1993, 2022-04-26:
[image]

> Also mind you, that you can't use that to exit if your PC hasn't enumerated the hid class report. In such cases, as always, the abort can be done via unplugging cable or VOL- & VOL+

Ah, that explains it - I've been primarily trying to cancel out of "waiting for connection" screen (I swear I've tested normal one too - but it apparently works now that I've double-checked). VOL- & VOL+ indeed works for that one. Sorry for the noise, it's slightly confusing.

CTCaer, 2022-04-26:
Yeah, joycon are not polled before the "Started hid emulation" message.
The VOL combo is always there to serve in such situations. (And I have it in anything in Nyx that disables on screen buttons and the procedure can take a long time).

Btw just to learn that console screen input debugging info:
[image]

That's the button bitfield. Each bit is a button (some bits are not button states, but anyway).

v1993, 2022-04-26:
Actually, I can reproduce the issue. It seems to happen after having hekate running for prolonged period of time. Restarting HID emulation does not get rid of it and in debug screen both buttons still register correctly.

CTCaer, 2022-04-26:
What issue and get rid of what.
I already told you how it works.
Anyway, I already tested the L3+Home abort and works **where it's expected to work** as it should.

closed 2022-04-26 by CTCaer

v1993, 2022-04-26:
Yeah, sorry for not being clear, I got what you mean. Checking again - for whatever reason, no matter what I try, L3 + HOME does not cancel out of "Started HID emulation" state (debug screen shows presses correctly and device is usable from PC side). Tried both cold boot and rebooting from Atmosphere - no apparent difference.

CTCaer, 2022-04-26:
Well that's not possible though. The code is simple:
https://github.com/CTCaer/hekate/blob/master/bdk/usb/usb_gadget_hid.c#L121-L123

The moment you see `Started HID emulation`, the joycon start to get polled.
That check there returns true if pressed and then the loop breaks and immediately shows `HID ended`.

The only way for that to not work, is if one of the joycon does not have good connection with the rail.
Because that depends on both of them being registered
If you remove the joycon, see that leds light up if you press a button and replug and it works, then it's a hw connection issue.
(Actual full polling starts, when both vibrate after a replug.)

v1993, 2022-04-27:
I think I've figured out what's going on (somewhat). If I poll joystick with something on PC side of things (evtest, yuzu, some SDL game, whatever), abort combination works just fine. But if I don't - it does nothing. I'm guessing it locks up somewhere in USB transfer?

Also made sure to check your suggestion about faulty joycon connection - alas, it's probably not that. Connecting joycon back always makes switch vibrate and does not affect if L3+HOME works or not. Besides, input works reliably when used as a controller.

CTCaer, 2022-04-27:
Usb transfers are in synced mode, so yeah, it stays there forever trying.
That was made with windows in mind which always reads it, because it has the interrupt for input devices always on.
You should have said that you are using linux, because that would clear up my confusion.
(I actually added the OR in the disconnect help message when I made that because of linux.)

All 3 major OS flavors actually behave totally different on usb stuff.
And unfortunately, the only stable behavior, can be found in windows.