CTCaer/hekate #741: error while loading fss0

ilykami, 2022-04-25:
cant load emunand due to 'atmosphere/fusee-secondary.bin wrong ini cfg or missing/corrupt files'

urherenow, 2022-04-25:
Stop using guides and download the files yourself from the source, and read the manuals while you're at it.

Your Atmosphere files are out of date, as there is no fusee anymore.

ilykami, 2022-04-25:
All my files are up to date, I reinstalled everything on my sd and it was fine until this morning

urherenow, 2022-04-25:
Lol... no they aren't. There is NO fusee-primary or fusee-secondary anymore. Only fusee.bin and package3.

impeeza, 2022-04-25:
Hello, this is no a support site, please go to proper sites like GBATemp or Discord.

Always READ the OP and readme.md before ask for help, if your read IT, you will find what you are using wrong config files.  Since September 6th of 2021, seven months now, the Atmosphere's file fusee-secondary was renamed to package3, so then you did should modify your hekate_ipl.ini file to reflect that change.

if this morning you have a atmosphere less of 1.0.0 and you did update to newer one, you need to edit your hekate_ipl.ini.

CTCaer, 2022-04-26:
From release that you didn't read:
___
[image]
___

And that notice btw, I'm supposed to remove because it's almost a year now.

closed 2022-04-26 by CTCaer

ilykami, 2022-04-26:
I know those files done exist, all my files are current and it was all working but after 12 hours it just stopped when I booted from rcm

ilykami, 2022-04-26:
Oh now it's saying failed to apply nosigchk

impeeza, 2022-04-26:
Support not here