CTCaer/hekate #743: [Feature Request] EmuMMC Config INI

Bladen123, 2022-05-09:
Hello CTCaer

Would it be possible that for each EmuMMC partition for example RAW1 and RAW2 that they do not use the same emummc.ini but use their own .ini like emummc_raw1.ini and emummc_raw2.ini?

as far as I know you could then use in the eummmc_raw1.ini an own version of atmosphere with the so-called atmosphere_path=emuMMC/RAW1/atmosphere or ? and so we could have 2 different CFW like an atmosphere standard without much garbage and without overclocking (p.s good for my girlfriend since she misclicked and overclocked my switch to 2000+ MHz on CPU i use the OC Suite from KazushiMe) and on RAW2 an atmosphere with own sysmodules , tesla , OC Suite etc .... on a SD without that they overwrite each other.

Masamune3210, 2022-05-10:
Isn't that the point of the emummc id?

CTCaer, 2022-05-10:
Yeah that was the point of emummc id. That was the original intention at least (and not the random dns usage).
But atmosphere never implemented contents per emummc id.

Anyway, that's the wrong repo to ask that. hekate and emummc driver already have support.

closed 2022-05-10 by CTCaer