CTCaer/hekate #747: Nyx submenu color/theming

Inclsevd, 2022-05-22:
Hello CTCaer,

I wanted to see what your thoughts would be on adding some very basic options for submenu theming in Nyx. Currently the Nyx background is not applied in submenus, I don't know your exact reason for that but I would guess it's to ensure clarity? That would make complete sense.

Specifically I would be interested in an option to switch to pure black background throughout to take advantage of the Switch OLED display, the hue graphics looks quite beautiful against pure black. I'm sure you're aware that OLED themes are quite common for various smartphone apps. Perhaps you could offer grey (default), black and white options?

Many thanks for reading (and also for the 10/10 bootloader)

CTCaer, 2022-05-27:
Clarity and performance reasons, as alpha blending is expensive.
Clarity is also important, especially when user needs to report an issue or something.
And of course it's a major PITA to add such support everywhere (I have to change every single style into a translucent one and then make it look nice).

For general black bg though, I might consider it because it looks nice yeah. Especially on that very bad OLED panel which has green pixel light uniformity issues on grays.
And it's easier to implement I guess than white (even if I actually have a white theme internally).

We'll see.

Inclsevd, 2022-10-12:
Very cool, thank you.

BeckysFootSlave, 2022-10-12:
> Many thanks for reading (and also for the 10/10 bootloader)

Just that one thing is not correct!
Sorry but it's a 16/10 bootloader 🥰😘