CTCaer/hekate #748: Cannot boot after emuMMC raw partition restore: PKG1/PKG2 mismatch - could not launch HOS!

ThinkpadT530user, 2022-05-27:
Hi, recently I did a raw partition backup since I wanted to add android to my switch. I have failed, but that doesn't matter. So I had to repartition my card which included wiping everything (I backed it all up), but the problem is that I have restored the emuMMC and the boot 0/1 (with raw partition on), but all I could get was Pkg 1 / Pkg 2 mismatch, could not launch HOS. I looked on google, but I couldn't,t find anything like this. Any help will be appreciated!

CTCaer, 2022-05-27:
Are you sure that you did boot0/1 AND raw gpp backups from emuMMC and you didn't mix them?
That error happens when pkg2 can't be decrypted.

That can happen if pkg1 does not match pkg2 master key version or if pkg2 is corrupted.
It's easy to fix if you know the HOS version btw. You can generate a BOOT0 (that has pkg1 inside) and flash it into emummc (with the raw partition button that you used).

ThinkpadT530user, 2022-05-27:
I'm pretty sure I didn't mix them, but you can be never sure. So how can you fix it?

CTCaer, 2022-05-28:
Well not the place to discuss that actually.
I told you when that the error "Pkg2 decryption failed! pkg1/pkg2 mismatch or old hekate!" can happen.

So you will need to search for boot0 creation/unbrick/whatever guide or seek support on a forum or something.

closed 2022-05-28 by CTCaer