CTCaer/hekate #749: Unknown pkg1 version. HOS version not supported! Or emuMMC corrupt!

ThinkpadT530user, 2022-05-28:
Hi again, after supposedly fixing my boot 0 and pkg2 error, I am greeted with a completely new problem. And now all it says is Unknown pkg1 version. HOS version not supported Or emuMMC corrupt! Trying backup bootloader... Found pkg1 <''>. Unknown pkg1 version. HOS version not supported! Or emuMMC corrupt! Failed to launch HOS! Press any key...

impeeza, 2022-05-28:
Buddy. Please seek for help on a proper site. This space is for bug report.

On gbatemp.net you are welcome to ask for help. You also can go to discord or similar site.

Regards

CTCaer, 2022-05-29:
Did you try to flash it on emuMMC (big button at the bottom in backup/restore screen enables emummc support)?
If yes, get a screenshot of console info -> microsd

ThinkpadT530user, 2022-05-29:
i did, still didn't work so i just migrated the sysmmc to emummc and started again.

closed 2022-05-29 by ThinkpadT530user