CTCaer/hekate #751: hekate freezes randomly

gusmmm, 2022-05-29:
after entering RCM and injecting hekate on a switch v1 firmware 14.1.1, hekate freezes randomly - sometimes imediately, sometimes after browsing some of the menus, sometimes while making a backup of the system. how can I check what's wrong? thanks!

CTCaer, 2022-05-29:
edit bootloader/nyx.ini `bpmpclock=1` to `bpmpclock=2`. Save and safely eject and reload nyx and tell me what happens.

masagrator, 2022-05-31:
Well, would be nice to have a pic of this error.

CTCaer, 2022-05-31:
@gusmmm you should contact me in discord in order to save time because that seems like a hw issue.
(you can join retronx or switchroot and then send a pm to me)

CTCaer, 2022-06-05:
User threw piracy stuff in the mix so no support will be given.

closed 2022-06-05 by CTCaer

gusmmm, 2022-06-05:
Thank you for your time.