CTCaer/hekate #753: Used Hekate emummc partition tool, and it wiped and corrupted the other partition.

CypherVault, 2022-06-16:
HI, ill try to be as detailed as possible, im besides myself right now. 

I used the hekate emummc oartition tool, gave the partition 29 gb, and then ran the setup in that partition, then went to use USB tools on the switch, and my real partition said F:/ cannot be accessed without formatting. Now i cant access anything on that part of the sd card. someone please help as i thought it would leave my files alone? i have not disturbed the partiiton since then someone please help.

CypherVault, 2022-06-16:
here is photo:
[image]

Masamune3210, 2022-06-16:
Open up disk management and take a screenshot of your disks, post it here please

CypherVault, 2022-06-16:
[image]
i read again the notes at the bottom of the screen it formats the user partition and i didnt use a save transfer tool.

Masamune3210, 2022-06-16:
According to explorer and disk management in the photos you sent, your fatè2 switch SD partition is mounted as D:\, not F:\

CypherVault, 2022-06-16:
correct. i thought that was my original partition, but it turns out hekate just formatted my ENTIRE user partition deleting all my user data

CTCaer, 2022-06-16:
User skipping 3 warnings and a full on time limited warning for button press, is not an issue of the app.

And btw the F:/ drive there is emuMMC and you can't access it. Also save files live in eMMC or in emuMMC partition. Not in the FAT one. FAT one only has game data which can be redownloaded in OFW.

closed 2022-06-16 by CTCaer