CTCaer/hekate #757: [Question] Do updates to EmuMMC with new HOS version support require migration

ChaosNicro, 2022-06-26:
Heya and thanks for providing this project.

In the changelog for V.5.7.2 under "EmuMMC Changes" it mentions new HOS 14.0.0 support and since I had newly created my EmuMMC with 5.7.0 and HOS 13.2.1, I wanted to ask:
Does the support require that a previous EmuMMC be migrated/repaired or even recreated before updating to 14.0.0
or does it simply mean that 5.7.2 should replace 5.7.0 as the bootloader before updating to 14.0.0 on the existing EmuMMC?

masagrator, 2022-06-26:
5.7.2 should replace 5.7.0 before update

Beside Issues is not for support. It's for bugs

ChaosNicro, 2022-06-28:
Sorry, I didn't see any channel for asking questions linked anywhere.

closed 2022-06-28 by ChaosNicro