CTCaer/hekate #759: Suggestion : Please create a new button to get to OFW

sthetix, 2022-07-02:
Hello, can you make a shortcut or a way to add to the launch menu to get to the real OFW ~ it is like pressing the Reboot button under Hekate then OFW. 

 I know there is a config called stock=1 under hekate_ipl.ini but I prefer to load the full stock without any patches 

Thanks.

impeeza, 2022-07-02:
I really don't understand well your query, but the "shortcut" for OFW can be created on the hekate_ipl.ini file, read the documentation and example.

sthetix, 2022-07-02:
When you read the documentation, creating the ofw shortcut under hekate_ipl
is not the real stock, but semi stock.

impeeza, 2022-07-02:
[image]
this is the most "Stock" you will get after booting hekate.

If you need Real deal, real stock, the only way is TO NOT BOOT HEKATE NOR ATMOSPHERE, but, you will burnt fuses, a possible ban and could get some others issues. because for that to work you need a clean SYSNAND and for that you has to read a lot of guides and documentation.

sthetix, 2022-07-02:
I know about fuses and I don't care about it. I mean I need to have a
button that don't bypasses the fuse count like the "semi" stock option
under hekate.

Masamune3210, 2022-07-02:
There is no point to a button existing to do something that would take a full reboot anyway. To boot fully stock, as already stated, you just boot normally without using rcm or a payload

sthetix, 2022-07-02:
I am talking about chipped consoles, 😊

pixeltester, 2022-07-02:
Why do you need a shortcut for two button presses?

sthetix, 2022-07-02:
Because One is better than Two

pixeltester, 2022-07-02:
So bloat the menu with a new radio button because you don't want to hit reboot and then OFW?

sthetix, 2022-07-02:
Of course not
There are plenty of option IF you want to think about it
Like for example CTCaer added an option to add 5 icons in a line than the
default 4 on the Launch menu
You can think about it IF you are creative enough.

pixeltester, 2022-07-02:
5 icons per line is not comparable to this request at all... Most users don't ever want to load OFW, the submenu protects against accidentally loading it.

sthetix, 2022-07-02:
The thing is not all users have the same perception. for me, I don't care
about OFW
but some people do

pixeltester, 2022-07-02:
If they care they can hit two buttons, this is a non request.

You say to think creatively but I say to think logically...

sthetix, 2022-07-02:
I suggest you think differently as well

Masamune3210, 2022-07-02:
There is still no point, the button already exists. If you cant be bothered to press two things and want to press just one instead, I don't know what to tell you

CTCaer, 2022-07-07:
stock does 0 patches.
I will remind again that stock **is** ofw (with hekate as bootloader and exosphere as secmon).
There's 0 actual configuration difference at boot/secmon/kernel/userspace levels.

That request comes and goes and every time it's asked for the wrong reasons.

So for the reasons stated here, as in other issues before it, again my answer is *use stock*.

closed 2022-07-07 by CTCaer

urherenow, 2022-07-12:
(sorry for the late comment... out to sea and haven't had a connection in a couple of weeks)

I might also add that any of the "upgradeable" chips with newer versions of hwfly-nx already can boot OFW by holding vol+ and vol- while powering on. No payloads are loaded this way.