CTCaer/hekate #761: hekate issue

Lofiif, 2022-07-06:
With a friend that had no problem using no problem with hekate, we tried on my switch but I have a graphic issue when I launch hekate.
I did every last updates he told me to do but it did not fix it. Here's what it looks like, I hope it can be fix. Also putting my sd card data if needed.
[image]
[image]

impeeza, 2022-07-07:
You need to delete your bootloader folder on SD card (backup it first) and then create a new one uncompressing the downloaded latest release of Hekate. You can restore HELATE_IPL.INI file but ONLY that file,. If you use them, reinstall sigpatches

impeeza, 2022-07-07:
Normally this is a res.pak damaged file.

CTCaer, 2022-07-07:
Yeah seems like res.pak is corrupted or missing
Backing up hekate_ipl.ini and deleting bootloader folder and pasting anew and restoring hekate_ipl.ini is the way to go.

closed 2022-07-07 by CTCaer