CTCaer/hekate #762: Hekate access impossible

armandgillot, 2022-07-06:
Hello,

I activated autocm but since the switch boot automatically on my emuMMC.. I don't know how to access hekate to deactivate autocm

Thank you for your help

Masamune3210, 2022-07-06:
hit one of the volume keys as soon as you see the hekate splash screen. next time ask on gbatemp or something please, this is more for issues than support

impeeza, 2022-07-07:
Hi. This is not a support forum. Need time please go to a proper support site. You need keep pressed Vol - while you inject hekate Payload to get in NYX there you can disable AutoRCM. Or better, edit your hekate_ipl.ini file to select what configuration you boot by default and the time hekate waits for you press Vol - to enter on NYX

closed 2022-07-07 by CTCaer

armandgillot, 2022-07-07:
Oops, I'm sorry,
But thanks for this information I was able to access hekate!

Have a good day