CTCaer/hekate #763: [bug] physical button navigation doesn't close prompts

monstrosities, 2022-07-09:
There's a minor issue with physical button navigation where going back doesn't close prompts but transitions the screen in the background. Maybe not actually a bug but a small oversight.

Example:

Step 1 - Disconnect UMS:

[image]

Step 2 - Press the physical back button "B", instead of the prompt being closed the screen transitions back in the background:

[image]

It also happens with partition manager prompts and probably other prompts.

CTCaer, 2022-07-10:
Well not a bug exactly, but limited usage of that feature.
The main cause is also the reason I don't have normal button support.

For making this work, I will need to have a list of button history and it's annoying to keep track of.
Maybe eventually when I will remove virtual mouse and add proper gamepad support, this will go away.
Until then it will work like that (only on windows Close button and not pop ups).

closed 2022-07-10 by CTCaer

monstrosities, 2022-07-10:
Ok, it was more of a "you might have missed this" type of thing rather than a real problem but the explanation makes sense.