CTCaer/hekate #765: hekate v5.8 emuMMC freezes

fecent, 2022-07-19:
If emuMMC has already been created, clicking the emuMMC button will freeze. When emuMMC is not created, it will freeze after clicking back after successful creation. The created emuMMC is used normally. This problem occurs in hekate v5.8, the device is OLED, before hekate v5.7.2 everything works fine

CTCaer, 2022-07-19:
What type of emummc? Also what happens if you rename /emuMMC/emummc.ini to .bak?

fecent, 2022-07-21:
> What type of emummc? Also what happens if you rename /emuMMC/emummc.ini to .bak?

Change to show not created state

CTCaer, 2022-07-21:
your emummc.ini is probably somehow corrupt or something.
Try enabling it from the change emummc menu and tell me if it hangs again.

fecent, 2022-07-22:
> your emummc.ini is probably somehow corrupt or something. Try enabling it from the change emummc menu and tell me if it hangs again.

No, it was newly created by hekate, and forced shutdown and restart can enter the emummc system normally, but clicking the emuMMC button will freeze, other functions will not be affected, is it my problem? My machine and my friend's machine will

fecent, 2022-07-22:
> your emummc.ini is probably somehow corrupt or something. Try enabling it from the change emummc menu and tell me if it hangs again.

ini file contents:
[emummc]
enabled=1
sector=0x0
path=emuMMC/SD00
id=0x30304453
nintendo_path=emuMMC/SD00/Nintendo

fecent, 2022-07-22:
> your emummc.ini is probably somehow corrupt or something. Try enabling it from the change emummc menu and tell me if it hangs again.

Hangs after clicking OK in the "The emuMMC Configuration was saved to sd card!"，after a minute or two, there will be a green screen
[image]

CTCaer, 2022-07-28:
I can't recreate your issue at all. Show me microSD info from Console info.

johnkelvinliu, 2022-07-28:
I have the same problem with that white screen here with 5.8.0 hekate with OLED switch here.
Lexar PLAY 1TB microSD card.
hwfly firmware 0.71

If creates emuMMC by "SD files" hangs just after successfully creating the emuMMC.
If creates emuMMC by partition 58GB, it can be created normally. But after reboot and going to hekate>emuMMC again, it will be frozen immediately.

It seems that no probelm with if creating emuMMC by 4GB partition when using PNY 64GB microSD card. 

I have put all apps and Easy Mode installed ITotalJustice sigpatches inside the card first.
The hekate>emuMMC forzening problem happens also by using HARD MODE installed ITotalJustice sigpatches in last version of hekate.

The hekate>emuMMC forzening proble can be easily solved by placing the hekate_ipl.ini file to the bootloader>ini folder.

I don't know if it could cause any other further problem.
It would better be fixed by using the main boot entries normally---placing the hekate_ipl.ini file to the bootloader folder.

johnkelvinliu, 2022-07-28:
I have put the hekate_ipl.ini backed to the file to the bootloader folder.
disabled the cheat at startup
change no hekate boot logo, auto start ATM, auto shutdown after auto reboot.
Then everything alright again. I still don't know which one helps.

CTCaer, 2022-07-28:
...
piracy is not supported

closed 2022-07-28 by CTCaer

fecent, 2022-09-13:
#771