CTCaer/hekate #768: Freeze on Hekate startup

lelaso, 2022-07-25:
Hello all.

I just bought a vulnerable switch. I have another hacked one, so I just followed the same steps in order to use Hekate with Atmosphere. The problem is that, when injecting Hekate, I got some problems.

Before using Lockpick, Hekate stands on its logo. After using Lockpick, the system shows the date window, and then becomes frozen until I force the turn off. I can access Hekate when I take the SD out, and then load some payloads when puting it back.

I tried using the same SD in my older switch with any issues. I am able to load fusee.bin and everything works just fine, but I would like to make a backup just in case.

Am I doing something wrong? I would like to know it as it seems that something happens when using Nyx.

Thanks in advance

Masamune3210, 2022-07-25:
Please try going to discord or gbatemp first, Github issues arent really super well designed for this kind of support question

CTCaer, 2022-07-26:
@Masamune3210 don't

@lelaso if you rerun it, it doesn't run fine? Basically you retried?
Tell me the contents of /bootloader/nyx.ini

lelaso, 2022-07-26:
I tried tons of times, I can't give you a number. Everytime goes the same: Nyx freezes on the first screen after the logo.

The content is this one:

> [config]
> themecolor=201
> timeoff=21af6740
> homescreen=0
> verification=1
> umsemmcrw=0

I just checked it was not a SD error, it lets me use Atmosphere with sigpatches, and I just installed a game in the SD with no lag or issue.

CTCaer, 2022-07-26:
piracy not supported

closed 2022-07-26 by CTCaer

lelaso, 2022-07-26:
I'm sorry, I didn't express correctly. I was attempting to show that it does not seem an error because of the SD reader, as I read in other forums. I didn't specify the nature of the title I installed, as I installed my own cartridge.

I'm only begging for help to use Nyx in order to make a backup of my console.

Sorry for the misunderstand.