CTCaer/hekate #769: "Contact me!" displayed next to display panel model

grave35, 2022-07-31:
As the title says.

Model: AUO A062TAN02 (30950F)
ID: [30] 97 [0F]

CTCaer, 2022-07-31:
A new one? I guess you don't have access to the LCD's back side, correct?

What's peculiar is that nintendo used 95 in your cal0. You didn't repair your lcd panel, right?

grave35, 2022-07-31:
This is not a new device. The LCD came from the store as a spare part for repair. I don't have access to the back of the lcd.  I saw in the menu "Contact me!" and informed you.

CTCaer, 2022-08-01:
Makes sense now. You or the repair shop didn't get any pictures of the back at all?
I would help to have the actual model number.

CTCaer, 2022-08-01:
Well photos are not legible in that store.
The issue is that adding that info to hekate is a 2-way process. I report ID in hekate but the model is still unknown.
Unfortunately, no one ever contacted me with any unknown actual model (not just id).
I just basically find the models by researching google images and piecing the puzzle together.

Anyway, thanks.

closed 2022-08-01 by CTCaer

grave35, 2022-08-03:
The seller sent me this photo.
[image]

CTCaer, 2022-08-03:
Thanks. That actually helps a lot (assuming it's from the panel sold to you and not another one).

Although it makes things weird:
```
 * [30] 93 [0F]: AUO A062TAN00 (59.06A33.000)
 * [30] 94 [0F]: AUO A062TAN01 (59.06A33.001)
 * [30] 95 [0F]: AUO A062TAN02 (59.06A33.002)
 * [30] 97 [0F]: AUO A062TAN02 (59.06A33.002)
 * [30] XX [0F]: AUO A062TAN03 (59.06A33.003) [UNCONFIRMED ID]
 ```
 
Except if A062TAN03 aka 59.06A33.003 has an id of 0x98.
And I maybe the assumed A062TAN00  is actually a 01 model.

I will add that info into sources.