CTCaer/hekate #771: Hekate freezes when emuMMC option is selected

BlackRabbit22, 2022-08-14:
### How to reproduce
***
1. On a Mariko unit, launch Hekate.
    * Make sure you have a partition based emunand selected.
2. Select emuMMC from the main menu.
3. Hekate freezes.

### Device information
***
**Hekate version:**
hekate v5.8.0 & Nyx v1.3.0

**Horizon version:**
14.1.2

**Device type:**
Mariko

### Description
***
When no emunand is selected, emummc button works fine. However when an emunand is already selected, selecting the emummc button in the main menu freezes  Hekate.

This issue is happening on Mariko units but not on Erista units.
I am using a partition based emunand. 

My `emummc.ini` file is not corrupt either. I tried deleting and remaking my `emummc.ini` file but the same issue occurs.

CTCaer, 2022-08-14:
I don't have that issue in a mariko unit.
Paste emummc.ini contents here using
\`\`\`ini
contents here
\`\`\`

and also paste a screenshot of console info -> microsd.

BlackRabbit22, 2022-08-14:
### emummc.ini
***
```
[emummc]
enabled=1
sector=0x364f8000
path=emuMMC/RAW1
id=0x31574152
nintendo_path=emuMMC/RAW1/Nintendo
```

### Screenshot
***
[image]

BlackRabbit22, 2022-08-14:
I tested out hekate v5.7.2 with Nyx v1.2.2 and everything is working properly.

CTCaer, 2022-08-15:
Don't test with the whole package replaced.
Put latest back and only replace nyx.bin with v1.2.2 one and tell me what happens.

BlackRabbit22, 2022-08-15:
Tested out Hekate v5.8.0 with v1.2..2 nyx.bin and the freeze no longer occurs. Everything is working fine now, issue seems to be from Nyx v1.3.0.

CTCaer, 2022-08-16:
What switch you have (although I find it strange that it matters)?
I can't replicate that on lite and oled.

Anyway find me in retronx discord and pm me.

CTCaer, 2022-08-16:
So it seems to be a bug with ini parsing that I can't replicate for whatever reason.
Some changes I did recently and internally on parsing, seem to fix the issue based on user's report.
So next version will have it.

johnkelvinliu, 2022-09-14:
Only using easy mode to install latest iPad, not using hard mode, then everything works fine.

CTCaer, 2022-10-12:
Fixed in v.5.9.0

closed 2022-10-12 by CTCaer