CTCaer/hekate #772: Display panel : contact me !

Sorries, 2022-08-16:
Hello, 

I have recently modded a brand new lite. It says contact the Ctc in the display panel field . How could I do that.

Thanks

CTCaer, 2022-08-16:
You are already doing that.
So what id it says it has?

Sorries, 2022-08-17:
It says ID: 40 94 10

CTCaer, 2022-08-17:
So you have the elusive 0x1040 panel.
Did you see model or vendor or any other markings on the panel and the flex cables?

Sorries, 2022-08-18:
[image]
[image]
[image]

dram ID 7.00 | 7.00

CTCaer, 2022-08-18:
Thanks that helps a lot.
I found that panel in a youtube video but was not sure about its model id being 40 94 10.
I will add the info in next release.

Btw the dram id I was talking about is at the left side of the hw info. It's called "DRAM ID" and it's under "SKU".
It's different than the 7.00 one because it comes from fuses and also I have the vendor and model.
Your switch is quite new so telling me what "DRAM ID" you have will help.

It looks like that:
[image]

Sorries, 2022-08-18:
No problem

The DRAM ID: 19: Samsung K4U6E3S4AA-MGCL 4GB

closed 2022-08-18 by CTCaer