CTCaer/hekate #773: UMS is not reliable on large transfers

Martmists-GH, 2022-08-18:
To give some context, I'm on the Luminescent modding team for BDSP. We have channels users can open if they have issues with installing.

Unfortunately, about 90% of the reports we get are about issues when installing with UMS:
[image]
[image]

However, we also did some internal testing:
[image]

While we're not sure *why* this happens, it always happens over UMS. It's possible that this is due to a faulty cable or USB connector, but hekate doesn't seem to give any warnings or errors if this is the case.

To help with reproduction, you can find the relevant zip file [here on google drive](https://drive.google.com/file/d/1gusgST7O0thL-WU8nuTgmSuBUtnU_Cuj/view?usp=sharing), and test by installing the mod for Pokemon Brilliant Diamond version 1.3.0. Reaching language select should be sufficient to see if it worked.

masagrator, 2022-08-18:
Did you ask if they were using exfat or fat32?

CTCaer, 2022-08-18:
UMS is raw access which means the OS can properly manage the filesystem.
Which means **if you improperly set archive bits on folders**, you will **break these and everything in them**.

FTP is a fake/virtual access from client to host and does not support archive bits and it will not set them.

Archive bits is the biggest issue of an out of spec usage in HOS and it should have been common knowledge by now.

You have 2 options:
- Fix the incorrectly packaged zip (it has archive bits in everything in the romfs folders and the folder itself)
 You need to remove archive bits from every single folder, otherwise HOS will think these are files.
 As your zip currently is, HOS sees the exefs folder and its files and a romfs file as broken. **Nothing else.**
- Leave it as is and make Archive bit fixer tool as a mandatory step on every extract

closed 2022-08-18 by CTCaer

Martmists-GH, 2022-08-18:
Do you have (links to) any tools that can show improperly marked folders on PC before being zipped? (any OS works)

urherenow, 2022-08-18:
Windows explorer? Right-click a folder or file, hit the advanced button towards the bottom-third of the box, see a check box for "file is ready for archiving"... at least I think that sets the bit...

Masamune3210, 2022-08-18:
It does, and yeah, best way I've found short of a bespoke tool or script is to just select the root folder and toggle the checkbox and then hit apply, Windows should then iterate through all the contents and remove the bit if present on anything

CTCaer, 2022-08-18:
On windows you can just select the files or folders and unset the "folder is ready for archiving" as said.
On anything else you can use a terminal command.

Also on most packers like 7zip/winrar/etc you can also choose to discard attributes/dates/etc.
On every cmdline archiver you can also do that almost always.

impeeza, 2022-08-18:
or on a command line you can use the command:

`attrib +A PATH_TO_FILE` (you can use wildcards)

it's easy for multiple files, you can use the parameter /S to process files on subfolders, by example:

`attrib +A *.* /S /D`

will put the archive bit on all files (and folders) on the actual folder and all subfolders.

CTCaer, 2022-08-18:
`+` sets it which is the opposite of wanted.
`-` clears it. (aka `attrib -a`)