CTCaer/hekate #774: hekate exception occurred (LR 4000D4D0)

RubyRoid, 2022-08-23:
Hi, I modded the Switch OLED yesterday and it worked fine. I left it on charger for a night and today went to install a larger SD card. Got this message. Tried old SD card and got the same error.

Can’t boot the console.

Occasionally it would show “Payload file is missing” text in red when I press a button.

I've tried hekate from official repo, the kefir one and one that comes from sdsetup zip archive. Always getting the same result.

Here's what is displayed on the screen: https://imgur.com/a/EJggAb9/

Please help!

closed 2022-08-23 by CTCaer

CTCaer, 2022-08-24:
BTW, since I prefer replies to be sometimes informative, I forgot to say the following edge case:
If your sd and small board connector is fine, since you have oled, and because you have the issue in lp0/minerva, it could also be a RAM chip issue.

I guess I can validate that if you want, but you will need to reach me in retronx discord (pm me).

RubyRoid, 2022-08-24:
Hi, thanks for your reply.

Yeah, I don't think the issue is with the SD or the connector. I've tried it with known good freshly formatted FAT32 cards (four of them) and also disassembled the console and inspected the connector pins, all looks perfect.

I pm'd you in discord.

leofrachet11, 2025-04-17:
Did you ever fix this, I'm getting a very similar issue but my error code is LR 4000E7C0

impeeza, 2025-04-17:
> Did you ever fix this, I'm getting a very similar issue but my error code is LR 4000E7C0

yeah necrobumping a closed thread will not help a lot, even more because is very probably a hardware issue.

leofrachet11, 2025-04-17:
I was just wondering if and how they managed to fix it because they never said