CTCaer/hekate #778: Compilation Error with newest release

Kekskruemel, 2022-09-09:
Hi,

hekate v5.7.2 & Nyx v1.2.2 compiles without problems, but with hekate v5.8.0 & Nyx v1.3.0 I get following error:

`Building build/hekate/main.o
./bootloader/main.c: In function 'ipl_main':
./bootloader/main.c:1524:2: error: a label can only be part of a statement and a declaration is not a statement
 1524 |  u32 *fb = display_init_framebuffer_pitch();
      |  ^~~
./bootloader/main.c: At top level:
cc1: warning: unrecognized command line option '-Wno-stringop-overread'
Makefile:147: recipe for target 'build/hekate/main.o' failed
make: *** [build/hekate/main.o] Error 1
`

Thx for your work.

Kekskruemel, 2022-09-09:
changing

skip_lp0_minerva_config:

to

skip_lp0_minerva_config: ;

fixed the problem.

urherenow, 2022-09-09:
You shouldn't have to mess with any code. It has compiled fine from the start.

When is the last time you updated your tool chain?

Kekskruemel, 2022-09-09:
Yeah Problem was a outdated tool chain.

Thx for the Tipp.

closed 2022-09-09 by Kekskruemel