CTCaer/hekate #779: Enhanchement for themes

impeeza, 2022-09-11:
On Hekate 5.8.0 if you use a file icon with "HUE" format, named *_hue.bmp, it is displayed well and with the color for the interface but if THE SAME file is renamed `*_hue_nobox.bmp` then the hue color isn't used, the icon is show without the gray background but white.

[image]

On this example all icons are the special ones with "Hue" format, but the second one is named `payload_hue_nobox.bmp` I don't know if is the wanted behavior or a bug introduced with the new functionalities.

CTCaer, 2022-10-17:
Next version will have it. (If I don't forget again to look at open issues. ^^)

impeeza, 2022-10-17:
Thank you very much for your hard work

CTCaer, 2022-12-20:
Added in 1.5.0

closed 2022-12-20 by CTCaer

fooness, 2024-05-25:
@CTCaer @impeeza Could you maybe link to *any at all* documentation/example on how to create “32-bit BMP” with transparency, so probably *ARGB*. Sorry for posting a comment to this closed issue, but for the life of me I could not find any working solution or proper answer on how to create these images (on macOS).

I tried creating such a `_hue.bmp` file with Illustrator, Photoshop, InkScape, GIMP, and ImageMagick, but to no avail. Nyx only ever displays the default switch image in the launch menu, even though when opening *my file* and [`hekate/nyx/resources/icon_lakka_hue.bmp`](https://github.com/CTCaer/hekate/blob/master/nyx/resources/icon_lakka_hue.bmp) seeminly look the same in Photoshop and ImageMagick’s `idenfity -v`.

@impeeza May I ask you to please share the atmosphere `_hue.bmp` file which is displayed in your screenshot above?

Thank you very much!